MarlinFirmware/Marlin #12626: [2.0.x] - Question about travel speed: it's less than expected

MasterPIC, 2018-12-11:
Marlin bugfix-2.0.x up to date - Anet A8 - stock config taken from config examples folder

With
> G91
> G1 X100 F10000

max speed is lower than expected.
Increasing feedrate for travel moves (e.g. by M205 T200) makes max speed increasing.
A further increase of feedrate for travel moves (e.g. by M205 T300) makes max speed increasing further.

Is it the expected behavour?

J-charles-C, 2018-12-11:
I,

Check your acceleration & Maw federate : 

` Default Max Feed Rate (mm/s)
 * Override with M203
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_MAX_FEEDRATE          { **300, 300**, 20, 170, 170 }

/**
 * Default Max Acceleration (change/s) change = mm/s
 * (Maximum start speed for accelerated moves)
 * Override with M201
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_MAX_ACCELERATION      { **500, 500**, 100, 10000, 10000 }

/**
 * Default Acceleration (change/s) change = mm/s
 * Override with M204
 *
 *   M204 P    Acceleration
 *   M204 R    Retract Acceleration
 *   M204 T    Travel Acceleration
 */
#define DEFAULT_ACCELERATION           **1000**    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  3000    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   **1000**    // X, Y, Z acceleration for travel (non printing) moves
`

MasterPIC, 2018-12-11:
I see that max travel speed is kept constant (so that a trapezoidal shape, as function of time, of the travel speed can be noticed).
But travel speed is limited neither by M203 nor by M201 nor by M204 in my tests...

I see that **M205 Tx limits max (travel) speed**. So I **think** M205 Tx sets **max** travel speed.

M205 **T** argument is not reported in config files; I found it only from Marlin documentation and now I'm tempted to think documentation is wrong since, referring to the documentation, M205 Tx is used to set **minimum** feedrate for travel moves (units/s).

@thinkyhead Am I wrong?

MasterPIC, 2018-12-12:
I gave a look at source code.
I think that currently tuning these settings obliges to study planner source code because of incomplete documentation.

closed 2018-12-12 by MasterPIC