MarlinFirmware/Marlin #12631: Delta Anycubic Kossel G33 ignore X_PROBE_OFFSET_FROM_EXTRUDER Y_PROBE_OFFSET_FROM_EXTRUDER

Piranit, 2018-12-11:
Hi everyone
I have an Anycubic Kossel Linear Plus 3D printer
I am making improvements with two 5015 fans (for better bridges and etc)
also decided to make a calibration through the proximity sensor
[image]
[image]
[image]
[image]

I'm using Marlin 1.1.9
in my case **X_PROBE_OFFSET_FROM_EXTRUDER  = -29** and **Y_PROBE_OFFSET_FROM_EXTRUDER = -57**
I **already change DELTA_CALIBRATION_RADIUS so probe check only inside bed**

but I have a problem, **G33 calibration does not build mesh properly** 
because **ignored X_PROBE_OFFSET_FROM_EXTRUDER and Y_PROBE_OFFSET_FROM_EXTRUDER**
**I found where is G33 code logic and there are not taken into account X_PROBE_OFFSET_FROM_EXTRUDER and Y_PROBE_OFFSET_FROM_EXTRUDER**
so the **calibration is wrong**
[image]

**Z offset I can calibrate perfect**
but bed **mesh calibrated wrong**

I moved probe sensor closer to a nozzle
[image]

but unfortunately, it's doesn't help(
here example print after g33
[image]
bugfix brunch doesn't help

Can anybody help how make works G33 calibration with X_PROBE_OFFSET_FROM_EXTRUDER and Y_PROBE_OFFSET_FROM_EXTRUDER?

LVD-AC, 2018-12-15:
G33 assumes the bed is **flat** and the towers are **square** to the bed. Bed **tilt** is not corrected by G33 and needs to be corrected mechanically in the build of the printer. G33 calibrates misaligned end-stops, delta radius and a not perfect equidistant triangle between the towers (tower angles). And for such calibration one needs to probe the height at specific positions of the mechanism (carriages, arms and effector) so for a center probe the 3 carriages should be at the same distance from the end-stops and the effector in the center so no XY_PROBE_OFFSET_FROM_EXTRUDER may be taken into account.

LVD-AC, 2018-12-16:
G33 works also fine with eccentric probes. It probes with the effector in the center and a number of points where the effector is on a circle. It does not matter if the probe has an offset or not.

Piranit, 2018-12-16:
I'm not interested in different probe types
I need solution for my inductive proximity probe with offset from nozzle

boelle, 2019-02-21:
@Piranit problem fixed?

Piranit, 2019-02-22:
no

boelle, 2019-02-22:
@Piranit tried latest copy of bugfix 2.0?

Piranit, 2019-02-28:
no, I have already sold this printer...

boelle, 2019-02-28:
then maybe click the close button below?

Piranit, 2019-02-28:
no, I'd tell the customer to try the last bugfix 2.0
And then I'd decide if this issue is still actual
If it's not, I'll close it.

boelle, 2019-02-28:
oki, it will just be hard for you to test yourself, it was just a thought

Piranit, 2019-02-28:
@boelle but for now, can you show me where delta new calibration logic with x/y probe offset is?

closed 2019-05-16 by thinkyhead