MarlinFirmware/Marlin #12632: BIGTREETRECH SKR V1.1 32bit LPC1768 mcu

weed2all, 2018-12-12:
Good morning!another question/new board for marlin...
After a search on Google I cross upon this board that in my vision has identical footprint as a MKS GEN L board only new is that have an LPC 1768 mcu as a Sbase and Smoothieboard..
Can marlin run this board?

 https://s.click.aliexpress.com/e/ckQ3MWVS

gloomyandy, 2018-12-12:
Hmm looks like another variation on a smoothieboard. It is different from the MKS Sbase as it seems to be using the same SPI channel for the LCD and the LCD SD card (unlike the SBase which uses different channels and instead shares an SPI channel between the two SD card connectors). This same SPI channel is bought out on to the LCD/SPI/SD header and is likely to be used when talking to TMC SPI devices. This may cause problems with some LCD displays. I'm not sure if there is a good pins.h file for this configuration so it may need a new one.

weed2all, 2018-12-12:
Yeah...but in my opinion this is better because of swapable drivers...i know that you can add external drivers to Sbase...but is not the same thing...

gloomyandy, 2018-12-12:
I wasn't saying if it was better or worse, just pointing out that it is different from what is probably the closest match in terms of existing pins files and some of the changes that will probably be needed if anyone wants to get one working.

dot-bob, 2018-12-16:
I have one sitting here I plan on installing on a corexy. I just need to find time to get around to it.  It should be simple to get up and running.  If I get to it before someone else I will submit a pull request to add a new board type and a pins file.

weed2all, 2018-12-16:
That will be awsome!

bipsendk, 2018-12-20:
Also looking at this board (actually just ordered one). Having it supported, hopefully also with support for 12864 RepRap Full Graphics LCD and MksMini would be awesome ....
As far as I can see, some pins are not the same as on the Smoothieboard (source: [https://www.biqu.equipment/collections/control-board-kits/products/bigtreetech-skr-v1-1-motherboard-32-bit-smoothieboard-arm-cpu-control-board-open-source-for-tft3-5-printer-pannel])

bipsendk, 2018-12-20:
I just noticed, there are some minor issues regarding the picture with the pin assignments. Both EXP connectors are shown as EXP1 .... And 0.15 is apparently connected on both EXP1 and EXP2, and 0.16 only exists on the LCD/SD/SPI pin array (which I think is a bug in the picture)...

XneddyX, 2018-12-20:
Hello i orderd Dome one to.... I'm in contact with Thema Dealer and i ask him about the Problems .. i will Report You...

bipsendk, 2018-12-21:
I am not so familiar with the LCP1768 - but would it be possible to use e.g. the Y_MAX  (P1_26) as PWM output to control a BLtouch probe ?

bipsendk, 2018-12-21:
Seems like some work already is in progress:  #12689

hectori4502, 2018-12-23:
other good board for marlin...?
REMRAM 
https://youprintin3d.de/elektronik/open-source/1029/remram-v1.5-first-edition.html

XneddyX, 2018-12-24:
This is the answere from BIQU BQ: "Hello friend.
The pin out photo is right.
You can see this video.
It can work with 12864LCD.
https://www.facebook.com/echo.li.520562/videos/226299174934140/?t=37"

bipsendk, 2018-12-25:
> This is the answere from BIQU BQ: "Hello friend.
> The pin out photo is right.
> You can see this video.
> It can work with 12864LCD.

Well, checking with the pinout of the RepRap Discount Full Graphics LCD, it would mean that the LCD4 pin on EXP1 is connected to the SCK pin of EXP2.. 
Also, they pinout shown designates EXP1 and EXP2 on the PCB - but the descriptive text states 2 connectors named EXP1 - so some sort of typo definitively exists on their pinout reference...

I will check up on it once I receive the board, that I have ordered...

b0ssman, 2018-12-27:
According to the smoothieboard bltouch wiring you wire can wire the servo to 1.23, 2.4, 3.25 or 3.26
http://smoothieware.org/zprobe#bltouch-or-servo-retractable-touch-probe
on the SKR 3.25 and 3.26 are on the LCD/SD/SPI header so in theory it should work.

b0ssman, 2018-12-27:
a i didnt see that they are using 3.25 and 3.26 for the lcd as well.
but 1.26 is also hardware pwm capable.
http://smoothieware.org/pwm-capable

Nuttavoot, 2018-12-28:
> a i didnt see that they are using 3.25 and 3.26 for the lcd as well.
> but 1.26 is also hardware pwm capable.
> http://smoothieware.org/pwm-capable

That smoothie board in the above link uses LCP1769, will it be same as LCP1768 ?

gloomyandy, 2018-12-28:
The LPC1769 is just a carefully selected version of the LPC1768 that can run at a faster clock speed. So yes the pins are the same.

Nuttavoot, 2018-12-29:
> i just ordered a SKR board. the price is unbeatable

Keep us posted for progress.

talldonkey, 2019-01-02:
Check out this branch from BIQU directly addressing the SKR:

https://github.com/Msq001/Marlin-bugfix-2.0-biqu

Source:
https://www.facebook.com/groups/505736576548648/permalink/604785386643766/

bipsendk, 2019-01-02:
Personally I prefer native support in the main branch - but as far as I can see from the commits, it is more or less already done. I am still waiting for my board, will test the bugfix-2.0 as soon as I can put it on the board (I have already tried to test-compile it in VScode without issues).

040medien, 2019-01-05:
I was able to get the latest Marlin 2.0 running on the SKR, but my REPRAP_DISCOUNT_SMART_CONTROLLER is not working. The backlight is coming on, but it's blank. It's not the contrast, so I'm wondering if the pins.h is correct for this combination. Did anyone manage to get the same combination to work?

gloomyandy, 2019-01-05:
@040medien it is not the same board but this one is pretty similar to the Sbase in many respects, so you may want to check out this thread (if you have not done so already) https://github.com/MarlinFirmware/Marlin/issues/12796

040medien, 2019-01-05:
Thanks for the pointer @gloomyandy - I made a mistake in my post - I'm actually trying to use the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER, but no luck so far. According to the post you linked to, this should work.

Miroslav57, 2019-01-06:
Hello 040 medien . I bought a skr v1.1 to and I have a reprap discount lcd from my Anet A6 but I can't made it work. The backlight isn't coming.

040medien, 2019-01-06:
Is that a standard RRD GLCD? If so, try flipping the cables 180 degrees. It's easiest if you carefully pull off the black plastic receptacles on either the board or the LCD.

Miroslav57, 2019-01-06:
Standard I thinks .It is a model 12864 with the scroll button and the reset button on the right of the screen . I already have turn th cable in.every way possible

kAdonis, 2019-01-06:
I used this display before
You need a custom cable or change the pins in the firmware, its not trivial.
You may consider buying a different display, the RRD GLCD is quite cheap

Miroslav57, 2019-01-06:
Ok thanks a lot . I will buying a new screen . Maybe an mks tft35

Miroslav57, 2019-01-06:
Ok so I have purchase a rrd glcd . Another questions did you succeed to install the driver for windows ? My computer didn't recognize the board

040medien, 2019-01-06:
Sorry, I'm using OS X.

gloomyandy, 2019-01-06:
What version of Windows do you have? The driver is already there for Windows 10. For other versions you can find it here: https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/Marlin/src/HAL/HAL_LPC1768/win_usb_driver

Others have reported that this driver works fine for them.

Miroslav57, 2019-01-06:
It is for windows seven

gloomyandy, 2019-01-06:
So did you try the driver I gave a link to? What happens?

Miroslav57, 2019-01-06:
No sorry tomorrow i will

myseg, 2019-01-08:
I have just received this board over the mail, and right after unboxing I realized what you were discussing here, that most of the pins are not broken out, they are just copies of aux1 and 2... Still, I am planning to hook it up with 3x tmc2130s and a bed leveling sensor, I will post some updates if I make any progress. If nothing helps, I might just hack some wires to the tiny pads of the microcontroller IC itself, as BIQU somehow failed to do so..
If I am mistaken, please correct me, I'd rather not hack into it unless I have to.

myseg, 2019-01-08:
Yes, that is what I am concerned about, I have no idea how it could possibly work but ill give it a try today. If this works, the tmc2130 could work like that on other boards too, making less wire bulk altogether. I don't have too high hopes though.
Still, it was 16 bucks, it's worth it to buy just for experiments

kAdonis, 2019-01-08:
>  they seem to have connected the drivers' CS pins to the enable pin next to each driver on the board.

Probably the drivers are enabled through SPI, so the EN pins are no longer needed, its a feature @teemuatlut is working on
AFAIK it is currently only supported for TMC2660 in mainstream Marlin, but its on the list for all smart drivers https://github.com/MarlinFirmware/Marlin/pull/11998#issue-220185771
@myseg it **could** work with Bigtreetech's Marlin fork, But that is just a guess

040medien, 2019-01-08:
I checked and Bigtreetech['s Marlin fork](https://github.com/Msq001/Marlin-bugfix-2.0-biqu) doesn't add more than pins_BIQU_SKR_V1.1.h - so it should work in mainline as well if it works at all.

myseg, 2019-01-08:
Also, here is the corrected pinout for it ( exp2 is upside down for better visualization)
[link](https://imgur.com/a/GEkv7KZ)

myseg, 2019-01-08:
Alright, got the sd card working, that's someting I guess

kAdonis, 2019-01-08:
The hardware SPI pins are:
```
  #define MISO_PIN            P0_17
  #define MOSI_PIN            P0_18
  #define SCK_PIN             P0_15
```
your MISO and MOSI pin definitions are swapped
Are you sure you see the external SD card and not the internal?

bipsendk, 2019-01-08:
FYI: It is a common problem, that some manufacturers rotate the IDC 10 pin connectors for EXP1 and EXP2 180 degrees.... I have also seen it on a MKS Gen v1.4 board (and using a standard RRD GLCD)

myseg, 2019-01-08:
Thank you all, I think I have figured it all out now. The geniuses at BIQU put the SCK pin (P0_15) to the lcd CS pin too (can be seen on the pinout too) no wonder it didn't work, ant that was why the sd worked  and the lcd did not (I figured out some time before my post that I mistakenly swapped the miso/mosi, the external SD definitely works). Now I need to hack that pin CS pin to the free P2.6 on the LCD/SD/SPI breakout. Will keep you posted tomorrow.

gloomyandy, 2019-01-08:
@myseg are you sure about the CS pin? Looking at this pinout:
https://ae01.alicdn.com/kf/HTB1NZmHaiDxK1RjSsphq6zHrpXac.jpg
it looks to me like pin P0_15 (SCK) and P0_16 (CS) are in the same place as on the MKS Sbase board (other than the SBase has the connector rotated) and that pinout works fine (at least it does with the RRD GLCD). P0_15(SCK) needs to go to both the SD card and the LCD.

What display are you actually trying to get working? The Graphics ones seem to work fine with the Sbase and other similar smoothie style boards but folks seem to have problems getting the simpler character ones (RRD SC https://reprap.org/wiki/RepRapDiscount_Smart_Controller) working. Perhaps you are running into the same issues?

b0ssman, 2019-01-09:
> I checked and Bigtreetech['s Marlin fork](https://github.com/Msq001/Marlin-bugfix-2.0-biqu) doesn't add more than pins_BIQU_SKR_V1.1.h - so it should work in mainline as well if it works at all.

its already merged

isnave, 2019-01-09:
hello I want to use the mks tft32, which I used in the mks gen l by plugging directly into aux-1, but on this board it does not work, it turns on but the commands do not work. someone could help me out please.

gloomyandy, 2019-01-09:
The tft32 is not really a display like the other more standard devices. It is more like having a separate small computer that talks to Marlin via a serial port and sends gcode commands to do things. This means you need to define a serial port to allow the tft32 to talk via. How have you configured the serial ports in your configuration file?

isnave, 2019-01-09:
I know it is the plate is 115000 baud and tft 250000 I did not remember ... what sweats

myseg, 2019-01-09:
> @myseg are you sure about the CS pin? Looking at this pinout:
> https://ae01.alicdn.com/kf/HTB1NZmHaiDxK1RjSsphq6zHrpXac.jpg
> it looks to me like pin P0_15 (SCK) and P0_16 (CS) are in the same place as [...]

So apparently, people at BIQU had no idea how the exp breakouts works, and they mapped the lcd CS pin (doglcd_cs) to the hardware SCK pin (P0_15) mistakenly. To access both the lcd and sd card, it needs two separate cs pins (DOGLCD_CS for the lcd, and SDSS for the SD) to work. The one SCK pin is enough, it only provides the clock, for both simultaneously.
This is not the same as a smoothieboard, the exp1 is not only rotated, it is completely messed up. I had to splice up the cable and remap the DOGLCD_CS pin to P2_06, and it finally works! See attached image. Now I will try the tmc2130s, hope they won't fry themselves...
This SKR v1.1 board is clearly a still at prototype stage, and we are the test subjects. Probably the v1.2 model will have correct pin breakouts. I really don't recommend anyone to buy it right now, unless they would want to use the tft32 screen (which I also do not recommend, the simple 12864 is far more useful). Better wait for a new version.

[image]

If I manage to hook up the tmc-s, I will post an update.

talldonkey, 2019-01-09:
Hi myseg, a BIQU employee seems to be actively designing the SKR 1.2 board at this FB thread/post, and has even included screens of the board/traces - I suggest you can provide that feedback directly to BIQU through this channel to ensure the same mistake isn't made and is actually fixed in SKR 1.2 .  https://www.facebook.com/groups/505736576548648/permalink/607735353015436/

myseg, 2019-01-10:
Wow! Did not see that yet. From the snaps he took from the board file, it looks like they broke out several more pins, and also managed to use the tmc2130 spi mode toggle jumpers, too, so no more wiring!
I guess we have the reason why this 1.1 board is so cheap, they were probably already working on the other ones

myseg, 2019-01-10:
Also, as the new board comes out in a short time, I think the thread can be closed. Everyone that had issues, thank you for beta testing for biqu for free (even we had to pay them), now buy the new, revamped board, it *might* not have issues...

(joke)

jmdearras, 2019-01-10:
> So apparently, people at BIQU had no idea how the exp breakouts works, and they mapped the lcd CS pin (doglcd_cs) to the hardware SCK pin (P0_15) mistakenly. To access both the lcd and sd card, it [...]

Does the cardreader work?  Can you share the config and wiring?

bipsendk, 2019-01-10:
@myseg - so with the definition above - and some rewiring of the LCD cables - it is possible to get the MKS_MINI_12864 to work - including the SDcard reader ?
It would be nice with a detailed description of which pins to connect where ...

myseg, 2019-01-10:
well, you turn the board around, there are the pin numbers. And to which pin goes where, it is in the code I posted just now. I have linked a pinout picture of both the mks12864, and the motherboard in this thread.

myseg, 2019-01-10:
Also, I have managed to hook up the tmc2130s, by using the X-min P1_29, and Y-min P1_27, and the second serial pins (P0_02 and P0_03, on the AUX-1 header) as CS pins (for 4pcs tmc2130-s). Connecting them to the enable pins leaded to erratic detection, but this way it works well. Shame that now I cant use the esp8266 wifi or other UART expansions though.
The other spi cables were simply connected onto the LCD/SD/SPI header pins, they are labeled.

Hope I helped everyone, after all it is not such a bad motherboard, it just lacks a lot of pins.

fape, 2019-01-11:
So hardware spi is working, good to know. I think we can use E1's en, dir and step pins to, besides unused endstops pins.  So you can free uart2 pins.

jmdearras, 2019-01-11:
Could you do a closeup of the wiring in the upper right corner?  I can *almost* make out all the connections.

jmdearras, 2019-01-11:
Could you post a close-up of the connections to the board?  The upper left portion of your picture.  I can almost make out the wiring changes.

Miroslav57, 2019-01-12:
Anyone could share his config.txt file for rrd glcd for smoothiware . The buzzer always ring when I switch on the board

[236 more replies]

closed 2019-03-06 by weed2all