MarlinFirmware/Marlin #12633: MKS Sbase v1.3 Bug

formu25, 2018-12-12:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2671727/Configuration.zip)
Hello 
Isn't there a problem with marlin2.0 latest version
Marlin 2.0 + mks sbase v1.3
The motors, the screen, don't work anymore?

gloomyandy, 2018-12-12:
Have you ever had that display work with your board? If not you may need to reverse one end of both cables.

formu25, 2018-12-13:
Advent the last build all worked, since you developed the support of TMC2130 rather than nothing worked

gloomyandy, 2018-12-13:
You are not telling us very much about the problem you have. The more information you provide the more likely it is that someone will be able to help.
* What version did you use before? Have you tried going back to that version, does it still work?
* Exactly what happens now, do you see anything on the display?
* What makes you think the problem you have has anything to do with TMC2130 support?
* How do you have your LCD connected to the board are you using standard cables or do you have a custom cable.
* What pins.h file are you using? Have you made any changes to it? If you have made changes please post a copy of your pins.h file.

Your configuration files seem to be based on an older version. For instance they do not contain the section that defines the type of stepper motor for each axis. I would suggest that you may want to start again with the default configuration for the Sbase and add your changes to it. 

I have the same board and display and they work fine. There have not been any other reports of problems with the same hardware that you are using.

formu25, 2018-12-13:
1: marlin 2.0 version about a month ago / yes with the previous version it still works
2: more screen display, more motor movement
3: just a guess, the problem and arrive since you have implemented the tmc 2130 function
4: I tested with several cables
5: no modification of the pin.h file
Okay, I'll reconfigure another marlin 2.0 file

formu25, 2018-12-13:
Problem locating 
When I'm with the DRV8825 it works
And when I replace the TMC2208 the problem appears
More engine and more display

gloomyandy, 2018-12-13:
You didn't mention anything in your original post about using TMC2208 drivers! So is that what you are doing? How do you have them connected? Are you using them standalone or are you trying to use a serial connection? If you are using a serial connection what pins are you trying to use?

formu25, 2018-12-13:
[image]

gloomyandy, 2018-12-13:
That looks like you are operating them in standalone mode. Have you defined that usage correctly in your configuration file? For X you need...
#define X_DRIVER_TYPE TMC2208_STANDALONE
Note that you must be using an up to date version of both the Configuration.h and Conficuration_adv.h files to be sure this will work correctly.

formu25, 2018-12-13:
Thanks, I'll go test it.

formu25, 2018-12-14:
thank you for your help in making it work

closed 2018-12-14 by formu25