MarlinFirmware/Marlin #12634: [bugfix-2.0.x] G29 J  broken in UBL

setar, 2018-12-12:
Hi 
i use current version (55144284c2578dccca7a6a5fa472a3d927ff09f2) of bugfix-2.0.x

I noticed that when you run several prints one after another, the printer’s table level calibration is disturbed

After a series of tests, I discovered that it was a function of the G29J
for example:
```
Connecting...
Printer is now online.
Using tool 0.
>>> g29 t
SENDING:G29 T
Bed Topography Report:
    (  1,389)                                      (389,389)
        0       1       2       3       4       5       6
 6 | +2.243  +2.193  +2.233  +2.161  +1.971  +1.938  +1.908
   |
 5 | +2.579  +2.523  +2.481  +2.334  +2.250  +2.204  +2.184
   |
 4 | +3.343  +3.225  +3.309  +3.205  +2.985  +2.912  +2.909
   |
 3 | +4.122  +3.837  +3.774 [+3.839] +3.744  +3.315  +3.547
   |
 2 | +4.455  +4.385  +4.248  +4.384  +4.254  +4.085  +4.066
   |
 1 | +5.039  +5.090  +5.024  +4.901  +4.867  +4.606  +4.652
   |
 0 | +5.624  +5.794  +5.800  +5.418  +5.481  +5.127  +5.238
        0       1       2       3       4       5       6
    (  1,  1)                                      (389,  1)
>>> g29 j v4
SENDING:G29 J V4
Bed X: 0.000 Y: 28.000 Z: 5.562
                Corrected_Z=0.56
Bed X: 390.000 Y: 28.000 Z: 5.454
                Corrected_Z=0.45
Bed X: 195.000 Y: 390.000 Z: 2.179
                Corrected_Z=-2.82
bed plane normal = [0.0002767,0.0091959,0.9999577]

>>> g29 t
SENDING:G29 T
Bed Topography Report:
    (  1,389)                                      (389,389)
        0       1       2       3       4       5       6
 6 | -0.515  -0.582  -0.560  -0.650  -0.859  -0.909  -0.957
   |
 5 | +0.416  +0.342  +0.282  +0.117  +0.015  -0.049  -0.087
   |
 4 | +1.774  +1.638  +1.705  +1.583  +1.345  +1.254  +1.233
   |
 3 | +3.148  +2.845  +2.765 [+2.811] +2.699  +2.252  +2.465
   |
 2 | +4.075  +3.988  +3.833  +3.951  +3.803  +3.616  +3.579
   |
 1 | +5.255  +5.287  +5.203  +5.063  +5.011  +4.732  +4.760
   |
 0 | +6.434  +6.586  +6.574  +6.174  +6.220  +5.848  +5.941
        0       1       2       3       4       5       6
    (  1,  1)                                      (389,  1)
```
as you can see from the debug, the first two points are very close to the virtual surface, and the third is significantly different - why?

I checked the tilt_mesh_based_on_probed_grid function in UBL_29.cpp and did not detect logical errors. To expand understanding, I put in an additional debug:
```
diff --git a/Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp b/Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp
index 57f21124b..5ec79f81a 100644
--- a/Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp
+++ b/Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp
@@ -1366,6 +1366,7 @@
           //z1 = measured_z;
           if (g29_verbose_level > 3) {
             serial_spaces(16);
+            SERIAL_ECHOLNPAIR("get_z_correction=", ubl.get_z_correction(PROBE_PT_1_X, PROBE_PT_1_Y));
             SERIAL_ECHOLNPAIR("Corrected_Z=", measured_z);
           }
           incremental_LSF(&lsf_results, PROBE_PT_1_X, PROBE_PT_1_Y, measured_z);
@@ -1380,6 +1381,7 @@
             measured_z -= get_z_correction(PROBE_PT_2_X, PROBE_PT_2_Y);
             if (g29_verbose_level > 3) {
               serial_spaces(16);
+              SERIAL_ECHOLNPAIR("get_z_correction=", ubl.get_z_correction(PROBE_PT_2_X, PROBE_PT_2_Y));
               SERIAL_ECHOLNPAIR("Corrected_Z=", measured_z);
             }
             incremental_LSF(&lsf_results, PROBE_PT_2_X, PROBE_PT_2_Y, measured_z);
@@ -1395,6 +1397,7 @@
             measured_z -= get_z_correction(PROBE_PT_3_X, PROBE_PT_3_Y);
             if (g29_verbose_level > 3) {
               serial_spaces(16);
+              SERIAL_ECHOLNPAIR("get_z_correction=", ubl.get_z_correction(PROBE_PT_3_X, PROBE_PT_3_Y));
               SERIAL_ECHOLNPAIR("Corrected_Z=", measured_z);
             }
             incremental_LSF(&lsf_results, PROBE_PT_3_X, PROBE_PT_3_Y, measured_z);

```

Here is the result I got:

```
Connecting...
Printer is now online.
Using tool 0.
>>> g29 t
SENDING:G29 T
Bed Topography Report:
    (  1,389)                                      (389,389)
        0       1       2       3       4       5       6
 6 | +4.712  +4.786  +4.861  +4.894  +4.761  +4.633  +4.617
   |
 5 | +5.281  +5.318  +5.256  +5.345  +5.107  +4.962  +4.897
   |
 4 | +6.129  +6.023  +6.082  +6.043  +5.888  +5.771  +5.679
   |
 3 | +6.813  +6.786  +6.729  +6.630  +6.633  +6.437  +6.376
   |
 2 | +7.338  +7.217  +7.212  +7.217  +7.142  +6.998  +6.925
   |
 1 | +7.914  +7.913  +7.904  +7.951  +7.816  +7.645  +7.700
   |
 0 |[+8.489] +8.610  +8.597  +8.685  +8.490  +8.292  +8.476
        0       1       2       3       4       5       6
    (  1,  1)                                      (389,  1)
>>> g28
SENDING:G28
>>> g29 j v4
SENDING:G29 J V4
Bed X: 0.000 Y: 28.000 Z: 5.543
                get_z_correction=5.00
Corrected_Z=0.54
Bed X: 390.000 Y: 28.000 Z: 5.504
                get_z_correction=5.00
Corrected_Z=0.50
Bed X: 195.000 Y: 390.000 Z: 2.217
                get_z_correction=5.00
Corrected_Z=-2.78
bed plane normal = [0.0000996,0.0091328,0.9999583]

Bed Topography Report:
    (  1,389)                                      (389,389)
        0       1       2       3       4       5       6
 6 | +1.957  +2.025  +2.094  +2.120  +1.981  +1.846  +1.824
   |
 5 | +3.117  +3.147  +3.080  +3.162  +2.917  +2.766  +2.695
   |
 4 | +4.555  +4.443  +4.495  +4.451  +4.289  +4.166  +4.067
   |
 3 | +5.830  +5.797  +5.733 [+5.627] +5.624  +5.422  +5.355
   |
 2 | +6.946  +6.818  +6.807  +6.805  +6.724  +6.574  +6.494
   |
 1 | +8.112  +8.105  +8.090  +8.130  +7.988  +7.812  +7.860
   |
 0 | +9.278  +9.392  +9.373  +9.455  +9.253  +9.049  +9.226
        0       1       2       3       4       5       6
    (  1,  1)                                      (389,  1)
```

It seems to me that this "get_z_correction=5.00" is not normal for all check points

This, for example, leads to the fact that three times the command G29 J significantly knocks the calibration of the table.

Please check others do the same?

Any idea?

gloomyandy, 2018-12-12:
You are probing points with the G29 J that are outside of your mesh (your mesh covers X=1,389, Y=1,389, but you are probing at 0 and 390). Try adjusting the points you probe to ensure that they fall inside of the mesh. One way to do this may be to expand your mesh to cover all of the bed by changing the value for MESH_INSET to 0. You have not provided your configuration files so it is hard to advise and further.

Oh and that mesh looks a little odd. Are you sure you have your z probe offsets set correctly? I would have expected to see a mesh centred around your homing position with both positive and negative offsets, your mesh is all positive.

alexxy, 2018-12-13:
But why G29 J touches data in bed matrix? isnt it supposed to be constant?

setar, 2018-12-13:
Thanks,
#define MESH_INSET 0  work fine for me.
All positive values for the table are protection against damage to the surface prior to calibration, or without switching on the auto level

setar, 2018-12-13:
A good idea would be to check if the G29 J measurement points are inside the grid.

gloomyandy, 2018-12-13:
I am not really sure how you have managed to configure things such that it probed outside of the grid. Is there a reason you are not posting your configuration files? 

@alexxy G29 J modifies the in memory copy of the mesh. It does this to adjust for any variation in home position or potentially for bed tilt (due to say a spring being compressed a little more). It allows you to have a large mesh without needing to reprobe all of the points, but still adjust for minor changes in bed setup. Of course you do not have to use it.

setar, 2018-12-14:
my config https://github.com/setar/Marlin/tree/bugfix-2.0.x-3ntProf_v10_AzteegX5_short_Yhead

Roxy-3D, 2018-12-15:
> @alexxy G29 J modifies the in memory copy of the mesh. It does this to adjust for any variation in home position or potentially for bed tilt (due to say a spring being compressed a little more). 

Yes.   Or in my case, printing on glass, and replacing the glass with a small piece of plastic ending up under the glass.   The mesh is valid,  but it has picked up a small tilt.

gloomyandy, 2018-12-15:
@setar can only see a couple of odd things. You have set MIN_PROBE_EDGE to 0 and you also have MANUAL_X_HOME_POS -25 which is outside of the limits for the bed you have set.
``` C
// The size of the print bed
#define X_BED_SIZE 390
#define Y_BED_SIZE 390

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 350
```

Can your probe really probe accurately at the very edge of your bed? Can you actually move to (-25, 0)?

As to having Marlin check that the probe M29 J probe points are inside the mesh, this is probably a good idea. But I'm not sure how exactly the location of points is determined for M29 J. @Roxy-3D would you have expected G29 J to have probed outside of the mesh?

Roxy-3D, 2018-12-15:
> would you have expected G29 J to have probed outside of the mesh?

The G29 probing logic assumes the specified points can be probed.    I don't think we want to add logic to the firmware to check the specified points make sense.   But we could add a 'Sanity Check' at compile time to insure the points make sense:

```
      if (do_3_pt_leveling) {
        measured_z = probe_pt(PROBE_PT_1_X, PROBE_PT_1_Y, PROBE_PT_RAISE, g29_verbose_level);
        if (isnan(measured_z))
          abort_flag = true;
        else {
          measured_z -= get_z_correction(PROBE_PT_1_X, PROBE_PT_1_Y);
          //z1 = measured_z;
          if (g29_verbose_level > 3) {
            serial_spaces(16);
            SERIAL_ECHOLNPAIR("Corrected_Z=", measured_z);
          }
          incremental_LSF(&lsf_results, PROBE_PT_1_X, PROBE_PT_1_Y, measured_z);
        }

        if (!abort_flag) {
          measured_z = probe_pt(PROBE_PT_2_X, PROBE_PT_2_Y, PROBE_PT_RAISE, g29_verbose_level);
          //z2 = measured_z;
          if (isnan(measured_z))
            abort_flag = true;
          else {
            measured_z -= get_z_correction(PROBE_PT_2_X, PROBE_PT_2_Y);
            if (g29_verbose_level > 3) {
              serial_spaces(16);
              SERIAL_ECHOLNPAIR("Corrected_Z=", measured_z);
            }
            incremental_LSF(&lsf_results, PROBE_PT_2_X, PROBE_PT_2_Y, measured_z);
          }
        }

        if (!abort_flag) {
          measured_z = probe_pt(PROBE_PT_3_X, PROBE_PT_3_Y, PROBE_PT_STOW, g29_verbose_level);
          //z3 = measured_z;
          if (isnan(measured_z))
            abort_flag = true;
          else {
            measured_z -= get_z_correction(PROBE_PT_3_X, PROBE_PT_3_Y);
            if (g29_verbose_level > 3) {
              serial_spaces(16);
              SERIAL_ECHOLNPAIR("Corrected_Z=", measured_z);
            }
            incremental_LSF(&lsf_results, PROBE_PT_3_X, PROBE_PT_3_Y, measured_z);
          }
        }
```

gloomyandy, 2018-12-15:
@Roxy-3D  Hmm in this case it is the default probe points (from Conditionals_post.h) that are outside of the mesh. It uses MIN_PROBE_X/MAX_PROBE_X etc. as the basis. But in this case some of those points do not fall within the mesh, because a combination of MIN_PROBE_EDGE set to 0 and MESH_INSET set to 1 results in them being just outside. Arguably this is a user error (since in this case @setar set MIN_PROBE_EDGE), but it would be nice if the default probe points could be chosen to take both the probable area (MIN_PROBE_EDGE etc.) and the area covered by the mesh (MESH_INSET etc.), but I'm not really sure of the best way to do that.

I guess that ideally we would have both a better default location being picked and also a check that the probe points are inside the mesh (in case the user has chosen the points explicitly and got them wrong).

setar, 2018-12-15:
> @setar can only see a couple of odd things. You have set MIN_PROBE_EDGE to 0 and you also have MANUAL_X_HOME_POS -25 which is outside of the limits for the bed you have set.

its my real geometry - long rail, and bed locate in middle zone.
Endstop locate in left edge for X. and back for Y

> Can your probe really probe accurately at the very edge of your bed? Can you actually move to (-25, 0)?

i can move head to (-25,0) but can not do probe - location out of bed surface

gloomyandy, 2018-12-15:
Yes but you currently have MANUAL_X_HOME_POS  (-25) set to be lower than X_MIN_POS (0) which is almost certainly wrong. If your machine can reach -25 then you should probably have X_MIN_POS set to -25.

setar, 2018-12-15:
X_MIN_POS = 0 need me for MIN_SOFTWARE_ENDSTOP_X
ok, i can set X_MIN_POS = -25
and set also #define MESH_MIN_X 0
right ?

gloomyandy, 2018-12-15:
You might want to spend some time looking at how the various coordinate systems (for the probe, the mesh and the bed) end up being setup. It can be complicated but is worth trying to understand especially if you have a machine that is unusual or has a bed that does not fill all of the possible movement.

From my reading of the code I don't think you need to set MESH_MIN_X it should be set to zero by default. From Conditionals_post.h...
```C
#define X_CENTER ((X_BED_SIZE) / 2)
#define X_MIN_BED (X_CENTER - (X_BED_SIZE) / 2)
#define _MESH_MIN_X (MAX(X_MIN_BED + MESH_INSET, X_MIN_POS))
#ifndef MESH_MIN_X
    #define MESH_MIN_X _MESH_MIN_X
#endif
```
Which is a very long winded way of setting MESH_MIN_X to 0 (assuming you have defined X_BED_SIZE and have MESH_INSET set to 0). But setting it should not do any harm.

setar, 2018-12-19:
Hmm :)

```
#define X_CENTER ((X_BED_SIZE) / 2)
#define X_MIN_BED (X_CENTER - (X_BED_SIZE) / 2)
```
Math logic tells me that this is the same:
```
#define X_CENTER (X_BED_SIZE) / 2)
#define X_MIN_BED 0
```
it does not depend on X_MIN_POS

closed 2018-12-19 by setar