MarlinFirmware/Marlin #12636: error trying to compile with bilinear ABL on Ramps 1.4

v1ru3, 2018-12-12:
Global variables use 16148 bytes (197%) of dynamic memory
seems like this is too big, i mean waaay too big.
What am I doing wrong?

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

gloomyandy, 2018-12-12:
perhaps if you posted your configuration files (as requested) someone will be able to tell you.

v1ru3, 2018-12-12:
wow... there you go:
/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H
#define CONFIGURATION_H_VERSION 010109

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// example_configurations/delta directory and customize for your machine.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a SCARA printer start with the configuration files in
// example_configurations/SCARA and customize for your machine.
[... 1835 lines cut ...]
#endif

/**
 * R/C SERVO support
 * Sponsored by TrinityLabs, Reworked by codexmas
 */

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Leave undefined or set to 0 to entirely disable the servo subsystem.
 */
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// Delay (in milliseconds) before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

#endif // CONFIGURATION_H

v1ru3, 2018-12-12:
this is what arduino says:

ketch uses 127502 bytes (50%) of program storage space. Maximum is 253952 bytes.

Global variables use 16148 bytes (197%) of dynamic memory, leaving -7956 bytes for local variables. Maximum is 8192 bytes.
Not enough memory; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing your footprint.
Error compiling for board Arduino/Genuino Mega or Mega 2560.

v1ru3, 2018-12-12:
> perhaps if you posted your configuration files (as requested) someone will be able to tell you.

i somehow feel like this is not new though #11747

Ludy87, 2018-12-13:
https://github.com/MarlinFirmware/Marlin/issues/11474 deals with the Anet Board

11474 deals with the Anet Board.
With your configuration I have determined these values:

```
DATA:    [===       ]  30.7% (used 2513 bytes from 8192 bytes)
PROGRAM: [==        ]  21.7% (used 55050 bytes from 253952 bytes)
```

Ludy87, 2018-12-13:
You can Close it yourself.

RudyBenner, 2018-12-13:
DATA:    [==========]  196.5% (used 16094 bytes from 8192 bytes)
PROGRAM: [=====     ]  53.2% (used 135174 bytes from 253952 bytes)

VSCode

RudyBenner, 2018-12-13:
Around line 625 in configuration_adv.h, things start to get funky.

v1ru3, 2018-12-13:
> Around line 625 in configuration_adv.h, things start to get funky.
> 50 works better in line 629.

:+1:

closed 2018-12-13 by v1ru3

fiveangle, 2018-12-13:
You used a Creality CR-10 configuration.  The CR-10 has 16k of RAM while the RAMPS has only 8k.

v1ru3, 2018-12-13:
> You used a Creality CR-10 configuration. The CR-10 has 16k of RAM while the RAMPS has only 8k.

you are partialy correct:

I used a CR-10 because I have a CR-1. However,  runs on ramps1.4 with DRV8825 and different lead screws (yes 2, one on each side ;) ) for Z.

fiveangle, 2018-12-15:
ibid