MarlinFirmware/Marlin #12637: Redundant sensor with 2 extruders and singlenozzle

JohnnyTheOne, 2018-12-13:
can I use redundant sensor with 2 extruders and singlenozzle?

I have this activated
```
#define EXTRUDERS 2
#define SINGLENOZZLE
```

If I activate
```
#define TEMP_SENSOR_1 5
#define TEMP_SENSOR_1_AS_REDUNDANT
```

I get this error message

```
In file included from Marlin\src\HAL\HAL_AVR\../../inc/MarlinConfig.h:41:0,
Cfompiling .pioenvs\megaatmega2560\src\src\HAL\shared\SpiEeprom.cpp.orom Marlin\src\HAL\HAL_AVR\HAL.cpp:29:

Marlin\src\HAL\HAL_AVR\../../inc/SanityCheck.h:1368:4: error: #error "TEMP_SENSOR_1 shouldn't be set with only 1 HOTEND."
#error "TEMP_SENSOR_1 shouldn't be set with only 1 HOTEND."

In file included from Marlin\src\HAL\HAL_AVR\../../inc/MarlinConfig.h:41:0,
from Marlin\src\HAL\HAL_AVR\HAL_spi_AVR.cpp:40:
Marlin\src\HAL\HAL_AVR\../../inc/SanityCheck.h:1368:4: error: #error "TEMP_SENSOR_1 shouldn't be set with only 1 HOTEND."
#error "TEMP_SENSOR_1 shouldn't be set with only 1 HOTEND."

In file included from Marlin\src\HAL\HAL_AVR\../../inc/MarlinConfig.h:41:0,
from Marlin\src\HAL\HAL_AVR\MarlinSerial.cpp:39:
Marlin\src\HAL\HAL_AVR\../../inc/SanityCheck.h:1368:4: error: #error "TEMP_SENSOR_1 shouldn't be set with only 1 HOTEND."
#error "TEMP_SENSOR_1 shouldn't be set with only 1 HOTEND."

In file included from Marlin\src\HAL\HAL_AVR\../../inc/MarlinConfig.h:41:0,
from Marlin\src\HAL\HAL_AVR\persistent_store_eeprom.cpp:24:
Marlin\src\HAL\HAL_AVR\../../inc/SanityCheck.h:1368:4: error: #error "TEMP_SENSOR_1 shouldn't be set with only 1 HOTEND."
#error "TEMP_SENSOR_1 shouldn't be set with only 1 HOTEND."

In file included from Marlin\src\HAL\HAL_AVR\../../inc/MarlinConfig.h:41:0,
from Marlin\src\HAL\HAL_AVR\servo_AVR.cpp:56:
Marlin\src\HAL\HAL_AVR\../../inc/SanityCheck.h:1368:4: error: #error "TEMP_SENSOR_1 shouldn't be set with only 1 HOTEND."
#error "TEMP_SENSOR_1 shouldn't be set with only 1 HOTEND."

*** [.pioenvs\megaatmega2560\src\src\HAL\HAL_AVR\HAL.cpp.o] Error 1
*** [.pioenvs\megaatmega2560\src\src\HAL\HAL_AVR\HAL_spi_AVR.cpp.o] Error 1
*** [.pioenvs\megaatmega2560\src\src\HAL\HAL_AVR\MarlinSerial.cpp.o] Error 1
*** [.pioenvs\megaatmega2560\src\src\HAL\HAL_AVR\persistent_store_eeprom.cpp.o] Error 1
In file included from Marlin\src\HAL\HAL_AVR\../../inc/MarlinConfig.h:41:0,
```

[Configuration.h](https://github.com/JohnnyTheOne/MarlinJTO/blob/JTO-A6/Marlin/Configuration.h)
[Configuration_adv.h](https://github.com/JohnnyTheOne/MarlinJTO/blob/JTO-A6/Marlin/Configuration_adv.h)

JohnnyTheOne, 2018-12-14:
I should have left #define TEMP_SENSOR_1 0 zero

closed 2018-12-14 by JohnnyTheOne

hopeinformer, 2019-08-03:
Thank you. This solved my problem as well.