MarlinFirmware/Marlin #12640: Thermal runaway on every print

Excaliburst, 2018-12-14:
I experience Thermal Runaway on every print, even though the wires are properly and safely connected with no room for shorts, and that the thermistor is placed in its hole in the heat-block.

My config
Printer: Anycubic Kossel Mini (Marlin FW 1.1.9)
It has earlier worked great with FW 1.1.0 RC.

I commence printing and it finishes the raft, but after a couple of layers of the actual print, it goes into Thermal Runaway, and needs to be reset.

What gives?

Thanks 

Morten E.

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

swilkens, 2018-12-14:
As requested in the standard text, please provide your `Configuration.h` and `Configuration_adv.h` files. You can zip them and drag the zip into the post window on github.

If you can, also try the current bugfix-2.0.x branch to see if the issue exists there as well, as that is where all future development is taking place.

Excaliburst, 2018-12-14:
My personal configurations - how do I make sure that they are easily copied into a new batch of firmware files? For instance - now you suggest me to try a new FW branch on the 2.0 branch. But this takes a lot of carefull copy pasting - or is there an easier way to swap FW... That I am missing.

I have attached my config files.

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/2679763/Configuration_adv.zip)

Thanks

RudyBenner, 2018-12-14:
Yes, it is a bit of work.
Have a look at this.
http://meldmerge.org/
I have had a look at your files, they seem to compile ok, but that does
not mean it will work. I do not like the settings on the temperature section.

RudyBenner, 2018-12-14:
Look for MELDMERGE

Excaliburst, 2018-12-14:
Thanks Rudy,

I have a diff-tool that shows me differences. But there are a lot of other diffs when comparing versions of same branch, and especially a newer branch file. 

The diff will not only show my edits to the file, but also all the needed code for the newer functionality. What a PITA... Sorry.

Excaliburst, 2018-12-14:
What don't you like about my temp section? What looks odd to you?

RudyBenner, 2018-12-14:
Do you have the FW 1.1.0 RC. files?

RudyBenner, 2018-12-14:
Another approach would be to start with the supplied example files for your printer.

RudyBenner, 2018-12-14:
also look here.

https://github.com/ruevs/Marlin_Anycubic_Kossel

Excaliburst, 2018-12-15:
Yes I have the 1.1. RC files. But they do not compile. I tested them yesterday.
I will attach them later today when I get to my PC.

I will have look at the Anycubic kossel files. Thanks Rudy.

Excaliburst, 2018-12-15:
Here are my Config files for 1.1.RC that worked in the past. I think I might have altered them later however. Not sure.

I'll have a look at your Github link. Thanks Ruby.
[Configuration_1.1.RC.zip](https://github.com/MarlinFirmware/Marlin/files/2682510/Configuration_1.1.RC.zip)

Excaliburst, 2018-12-15:
I have now tried to adjust the settings from the Github link above. But now get an error compiling the FW.

It seems the problem is with the board, but the setup is not changed from when I successfully flashed the printer.
Board: Arduino/Genuino Mega or Mega 2560
Processor: Atmega 2560 (Mega 2560)
COM port: 4
************

Gives me this error trying to compile
Arduino: 1.8.1 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\ultralcd.cpp: In function 'menu_action_setting_edit_callback_long5.constprop':

sketch\ultralcd.cpp:4332:3: internal compiler error: Segmentation fault

   DEFINE_MENU_EDIT_TYPE(uint32_t, long5, ftostr5rj, 0.01);

   ^

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: C:\Users\Morten\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/users/morten/appdata/local/arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "LiquidCrystal.h"
 Used: C:\Users\Morten\Documents\Arduino\libraries\LiquidCrystal
 Not used: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

boelle, 2019-02-21:
@Excaliburst still a problem?

boelle, 2019-02-25:
clogging nozzle can be many things

personal i only had extruder problems when i forget to hot tighten the nozzle like i should, the nozzle will then leak and the heatbreak gets filled with molten plastic

the thermal run-a-way, is it like it cant get hot enough? or does it get to hot?

thinkyhead, 2019-02-25:
Please test the latest `bugfix-1.1.x` and/or `bugfix-2.0.x` branches to see whether it helps. If the problem has been resolved then we can close this issue. If the issue isn't resolved in either of those branches yet, then we should investigate further.

Excaliburst, 2019-02-25:
@boelle Can you please explain the issue you had, and how to fix it. It sounds a lot like what I am experiencing here. Thanks Bo.

@thinkyhead I will try the lates bugfix-2.0.x branch. And return with my findings.
Is my printer (Anycubic Kossel Delta) compatible with 2.0, or should I stick with v. 1.1.x branch. What is the difference between them?

Thanks a lot guys

boelle, 2019-02-25:
so you cant keep it hot? or does it get to hot?

boelle, 2019-02-25:
2.0 is updated more than the old versions, and 2.0 is both 32 bit and 8 bit

Excaliburst, 2019-02-26:
@boelle I do not have any thermal runaway issues anymore. My issues now are solely clogging the hotend on every print... Please can you explain to me, how you fixed that issue with your setup?
Please be more detailed than the first post here.

"personal i only had extruder problems when i forget to hot tighten the nozzle like i should, the nozzle will then leak and the heatbreak gets filled with molten plastic"

Please explain how you fixed this issue. It seems similar to what I am experiencing. Thanks

boelle, 2019-02-26:
i cant be more detailed than that

Excaliburst, 2019-02-26:
What did you tighten at the hotend? I am not clear as to what can be adjusted - tightened... ?

Excaliburst, 2019-02-26:
When I try to compile the Marlin FW 2.0 Bugfix branch with an Anycubic Delta setup from example config. I get this error:

Arduino: 1.8.1 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Build options changed, rebuilding all
Multiple libraries were found for "LiquidCrystal.h"
 Used: C:\Users\Morten\Documents\Arduino\libraries\LiquidCrystal
 Not used: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal
fork/exec C:\Users\Morten\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2/bin/avr-gcc.exe: The filename or extension is too long.
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
*********

Help anybody?

boelle, 2019-02-26:
if you used seach in youtube you would find this: https://www.youtube.com/results?search_query=e3d+v6+assembly

boelle, 2019-02-26:
> When I try to compile the Marlin FW 2.0 Bugfix branch with an Anycubic Delta setup from example config. I get this error:
> 
> Multiple libraries were found for "LiquidCrystal.h"

there is the reason why it fails, you have more than one lib with the same name

google how to fix that

we are here to help you help yourself, not do the work for you

thinkyhead, 2019-03-03:
Actually, the reason for the error is Arduino 1.8.1. You need to update to 1.8.8 or later.

closed 2019-03-03 by thinkyhead