MarlinFirmware/Marlin #12641: Err mintemp bed printer halted using Marlin 1.1.9 on creality cr10 10

johanmga, 2018-12-14:
I use marlin 1.1.9 on cr 10 with melzi board 1.1.2 everything was fine but in the past days I got the mintemp bed error , I change the thermister and the problem still the same, so I look into the marlin code more and change the thermister type that are set 5 to 1 that I see that number 1 is the original value in the marlin stock firmware, I test again and the problem still, so I decide to change the marlin 1.1.9 and back to the stock firmware, do a print and everything ok, monitoring that via octoprint see a temp variation in the bed but the printer not halted in the stock firmware, what can be the problem here? What are the values that control the halting trigger on marlin 1.1.9? I will appreciate your help!!

swilkens, 2018-12-14:
If the issue started without changing the firmware, it's probably an electrical / mechanical issue - not a firmware problem.

MINTEMP usually indicates a disconnected thermistor, verify that you made the connections properly and verify that the wires leading up to the thermistor aren't broken. A multimeter of some kind should be fine for this work. If you can, measure the resistance of the thermistor. It should be around 100K.

Did you change your firmware when this started happening? If so, please attach your configuration files as indicated in the template when you start a new issue report.

johanmga, 2018-12-14:
@swilkens  i change the firmware for 1.1.9 before that the printer starts to show that behaviour, so i back to the stock firmware and the printer can finish the printer jobs but in octoprint i can view a serious temperature fluctuation in the bed, i dont think that is problem with the termister because i change the thermister for a new 100 k thermister, so i do the following tests:

1. change the 100uf 16 v capacitor in the melzi board for a new 470 uf 16v capacitor
2. do a new print and see the following behaviour [https://imgur.com/gallery/TK1w2nL]( https://imgur.com/gallery/TK1w2nL)

when i print with the 100uf capacitor the temperature fluctuates even more, but i can finish the prints with the stock firmware with the marlin 1.1.9 i can not finish the prints i think that is maybe in the stock firmware in configuration_adv.h  has differences in the termal settings specially this lines are commented

//// Heating sanity check:
// This waits for the watchperiod in milliseconds whenever an M104 or M109 increases the target temperature
// If the temperature has not increased at the end of that period, the target temperature is set to zero.
// It can be reset with another M104/M109. This check is also only triggered if the target temperature and the current temperature
//  differ by at least 2x WATCH_TEMP_INCREASE
//#define WATCH_TEMP_PERIOD 40000 //40 seconds
//#define WATCH_TEMP_INCREASE 10  //Heat up at least 10 degree in 20 seconds

but in configuration adv.h in marlin 1.1.9 that feature is enabled.
i appreciate if any of you can help me to fix the problem specially for use the more new marling 1.1.9 firmware.
thanks a lot!!

[image]
[image]
[image]
[image]
[image]
[image]
[image]

boelle, 2019-02-21:
@johanmga problem fixed?

johanmga, 2019-02-21:
@boelle I see that it's a hardware problem with the creality v 1.1.2 board that in stock does not "feel" well the temperature of the bed cause of a 5v voltage fluctuations, I plan to swap this board for an mks gen l but I'm waiting for the steppers , thanks a lot for the help

boelle, 2019-02-21:
@johanmga so its not a problem with marlin?

boelle, 2019-02-21:
maybe click the green close button below if its not marlin problem

closed 2019-02-21 by johanmga