MarlinFirmware/Marlin #12643: [1.1.9] Range of Auto Temp Factor is limited in LCD menu

Elump, 2018-12-14:
Hi,

I am wondering about Auto Temp Factor range limitation in LCD menu. Given range is 0 to 1.

Description:
To make Auto Temp useful, my printer needs factor >1 and I can't see any need to limit Auto Temp Factor to <=1.

Steps to Reproduce:
-activate Auto Temp 
-connect LCD with activate LCD menu
-change Auto Temp Factor in LCD menu

Expected behaviour:
reasonable range of Auto Temp Factor should be available in LCD menu

Actual behaviour:
Auto Temp Factor range is limited from 0 to 1.

Additional Information:
in ultralccd.cpp change 
`MENU_ITEM_EDIT(float52, MSG_FACTOR, &planner.autotemp_factor, 0, 1);`
to e.g.
`MENU_ITEM_EDIT(float52, MSG_FACTOR, &planner.autotemp_factor, 0, 10);`

boelle, 2019-02-21:
@Elump problem fixed?

Elump, 2019-02-22:
@boelle with following change the problem is fixed. But, I'm new to Git and can't do pull request...

in ultralccd.cpp change
<code>MENU_ITEM_EDIT(float52, MSG_FACTOR, &amp;planner.autotemp_factor, 0, 1);</code>
to e.g.
<code>MENU_ITEM_EDIT(float52, MSG_FACTOR, &amp;planner.autotemp_factor, 0, 10);</code>

boelle, 2019-02-22:
using bugfix 2.0?

AnHardt, 2019-02-22:
Please have a look on #13227 and #13226.
Is 10 enough?

commit 31b0be74 (2019-02-22): Fix autotemp_factor menu setting (#13226)

Fix #12643
Only factors above 1 can have a 'positive' effect.
closed 2019-02-23 by thinkyhead