MarlinFirmware/Marlin #12644: TMC2130 Shut down mid-print without current reduction

towe96, 2018-12-14:
### Description
Setup:
CoreXY printer, 1/16th microstepping, MKS GEN 1.4 with Marlin 2.0, 5x TMC2130 by Watterot in SPI on a 24V step-up with 100uF capacitors directly on the drivers

### Steps to Reproduce

1. Acceleration > 3000, Current (RMS) > 700
2. Printing something like infill with many direction changes
3. Steppers for X, Y and E shut down - commands keep being sent, but the drivers report "disabled" in M122. Sending "M906 X700 Y700" or any other SPI command "wakes them back up" - is this thermal shutdown?

**Expected behavior:** Drivers reduce current if too hot - behaviour changes proportionally to cooling efforts

**Actual behavior:** Drivers just shut down without any message or "fault" reported, temperature flag not set - it doesn't matter whether I blow on the drivers with a 40mm fan, 120mm fan or no fan at all.

Glod76, 2018-12-15:
Post configs for help

towe96, 2018-12-22:
Sadly this is still an issue for me.
Even at 3000 mm/s^2 acceleration and 775 mA, the drivers disengage after a longer print - in this case a lithophane. Speed is only 40 mm/s.
The drivers simply disengage, and the affected driver registers read 0xE0:00:00:00, while the working ones (Z and Z2) read 0x80:07:00:00.
There's no way it's a thermal issue - I now have both a 120mm and 40mm fan blowing at the heat sinks, and they never trigger the OT prewarning.

I'm just lost...

burk80, 2019-01-09:
alexsomesan commented on repetier forum.. (https://forum.repetier.com/discussion/comment/21465/#Comment_21465)

It is better to solder the spi pins on the bottom of the board instead of changing the pins on the top due to power lines F*ing up the spi data with the cable solutions..

Losing the connection to the 2130 board.

seems like this might be a similar problem

boelle, 2019-02-21:
@towe96 problem fixed?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead