MarlinFirmware/Marlin #12652: Thermistor is stuck at 0 when enabling dual steppers for Y-axis

eyewy, 2018-12-15:
### Description

I have a 3d printer with 2 extruders. I want to sacrifice the second extruder to use the 2nd filament driver for driving the 2nd stepper on Y-axis. So in configuration_adv.h (attached), I added:
#define Y_DUAL_STEPPER_DRIVERS
This correctly re-purpose the 2nd filament stepper driver for Y-axis, however, now the extruder 0 thermistor always reads 0. I tried putting the two config files in the latest Marlin version (1.1.x) folder, and compile, but the issue still persists. 

#### Additional Information
[config.zip](https://github.com/MarlinFirmware/Marlin/files/2683035/config.zip)

 `Configuration.h` and `Configuration_adv.h`  attached.

RudyBenner, 2018-12-17:
I see 

#define CONFIGURATION_H_VERSION 010109

in your config file.

boelle, 2019-02-21:
@eyewy problem fixed?

closed 2019-03-03 by thinkyhead