MarlinFirmware/Marlin #12658: [bugfix-2.0.x] 'change filament' is broken after bed leveling

lujji, 2018-12-16:
### Description
Changing the filament after leveling the bed results in Y axes travelling into infinity.

### Steps to Reproduce
1. Motion -> Level Corners
2. Change filament -> Preheat PLA
3. Z axis lifts up slightly and and Y axis crashes into the end-stop

[config.zip](https://github.com/MarlinFirmware/Marlin/files/2683600/config.zip)

boelle, 2019-02-21:
@lujji still a problem?

lujji, 2019-02-24:
I haven't tested the latest commit. So far I discovered that leveling corners somehow messes up XYZ coordinates, which are stored in global variables (why?). As a workaround I specify the exact coordinates with M600 command:

``` diff
inline PGM_P _change_filament_temp_command() {
       return _change_filament_temp_extruder >= 0 ? PSTR("M702 T%d") : PSTR("M702 ;%d");
     case ADVANCED_PAUSE_MODE_PAUSE_PRINT:
     default:
-      return PSTR("M600 B0 T%d");
+      return PSTR("M600 B0 T%d X10 Y15 Z5");
   }
   return PSTR(MSG_FILAMENTCHANGE);
 }
 
 static void _change_filament_temp(const uint16_t temperature) {
-  char cmd[11];
+  char cmd[32];
```

boelle, 2019-02-24:
@thinkyhead @Roxy-3D does this make sense to you?

thinkyhead, 2019-03-03:
Without additional data we can only speculate. For proper troubleshooting we need to see log output, particularly what `M114` says at various points, after using level corners and before invoking `M600`, then after invoking `M600` with no parameters, and so on.

boelle, 2019-03-06:
@thinkyhead i think we can close this one

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

lujji, 2019-06-23:
@boelle - I have no access to the machine so I can't test if this issue got resolved in the latest commit (judging by the same behavior I've seen on other machines running 2.0.x - it's not). Since you don't care about fixing bugs - feel free to close this issue.

lujji, 2019-06-30:
@thinkyhead I've finally managed to get my hands on the printer. Tested on 0ca64a0ea34f5bf96c4122b2f8341ccbcb0f6f2f, problem persists.
```
// M114
[22:05:14:317] X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
[22:05:14:317] ok

// level corners
[22:05:34:146] echo:busy: processing
[22:05:36:161] echo:busy: processing
[22:05:38:159] echo:busy: processing
[22:05:40:158] echo:busy: processing
[22:05:42:157] echo:busy: processing
[22:05:44:156] echo:busy: processing
[22:05:46:171] echo:busy: processing
[22:05:48:170] echo:busy: processing
[22:05:49:721] X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
[22:05:49:721] ok

// M114
[22:06:12:700] X:30.00 Y:30.00 Z:0.00 E:0.00 Count X:2400 Y:2400 Z:0
[22:06:12:700] ok

// change filament -> preheat PLA: Y axis goes insane
[22:06:31:937] echo:enqueueing "M600 B0 T0"
[22:06:33:936] echo:busy: processing
[22:06:35:951] echo:busy: processing
[22:06:37:950] echo:busy: processing
[22:06:38:388] X:10.00 Y:225.00 Z:20.00 E:0.00 Count X:800 Y:18000 Z:8000
[22:06:38:388]  T:31.20 /185.00 B:25.54 /0.00 @:127 B@:0 W:?
[22:06:39:379]  T:32.53 /185.00 B:25.51 /0.00 @:127 B@:0 W:?
[22:06:39:955] echo:busy: processing
[22:06:40:387]  T:33.70 /185.00 B:25.45 /0.00 @:127 B@:0 W:?
```
Executing "M600" manually with no parameters yields the same result.

shitcreek, 2019-08-07:
I am unable to reproduce this error. Are you still having this issue? I'm not sure if it matters but try enabling HOME_BEFORE_FILAMENT_CHANGE. Also you could enable M114_DETAIL for more details.

boelle, 2019-08-19:
@lujji tried what @shitcreek suggested?

lujji, 2019-08-27:
@shitcreek I did try enabling HOME_BEFORE_FILAMENT_CHANGE and had the same result. Could you post your config so I can diff it with mine?

boelle, 2019-10-24:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-24 by boelle