MarlinFirmware/Marlin #12659: [bugfix-2.0.x] #define DELTA_AUTO_CALIBRATION Issue

lkvapil, 2018-12-16:
`Marlin/src/lcd/menu/menu_delta_calibrate.cpp: In function 'void _man_probe_pt(const float&, const float&)':
Marlin/src/lcd/menu/menu_delta_calibrate.cpp:42:76: error: 'DEFAULT_XYZ_STEPS_PER_UNIT' was not declared in this scope
move_menu_scale = MAX(PROBE_MANUALLY_STEP, MIN_STEPS_PER_SEGMENT / float(DEFAULT_XYZ_STEPS_PER_UNIT));
^~~~~~~~~~~~~~~~~~~~~~~~~~
Marlin/src/lcd/menu/menu_delta_calibrate.cpp:42:76: note: suggested alternative: 'DEFAULT_AXIS_STEPS_PER_UNIT'
move_menu_scale = MAX(PROBE_MANUALLY_STEP, MIN_STEPS_PER_SEGMENT / float(DEFAULT_XYZ_STEPS_PER_UNIT));
^~~~~~~~~~~~~~~~~~~~~~~~~~
DEFAULT_AXIS_STEPS_PER_UNIT
*** [.pioenvs/LPC1768/src/src/lcd/menu/menu_delta_calibrate.cpp.o] Error 1`

boelle, 2019-02-21:
@lkvapil problem solved?

thinkyhead, 2019-03-03:
See example configs. Add to `Configuration.h`:
```cpp
#define DEFAULT_XYZ_STEPS_PER_UNIT ((XYZ_FULL_STEPS_PER_ROTATION) * (XYZ_MICROSTEPS) / double(XYZ_BELT_PITCH) / double(XYZ_PULLEY_TEETH)) // 80
```

closed 2019-03-03 by thinkyhead