MarlinFirmware/Marlin #12665: [2.0.x-bugfix] mks sbase does not use or save PID values

alexzapatka, 2018-12-17:
### Description
when trying to use custom pid values either from eeprom or programed in to configuration.h, my board is ignoring them

#### Additional Information

i have an MKS SBASE 1.3.  i have compiled marlin for my sbase and it seams to work well except it is ignoring PID values.  i used the pid auto tune to come up with values for both my hotend and my build plate and it works fine.  

i get (from octoprint) 
```
Recv:     #define DEFAULT_Kp 15.1
Recv:     #define DEFAULT_Ki 1.24
Recv:     #define DEFAULT_Kd 46.38
```
and 
```
Recv:     #define DEFAULT_bedKp 179.5
Recv:     #define DEFAULT_bedKi 12.28
Recv:     #define DEFAULT_bedKd 656.64

```
which i then add to my Configuration.h and recompile and reinstall the firmware.  

but when i reconnect to it i get:

```
Recv: echo:PID settings:
Recv: echo:  M301 P22.20 I1.08 D114.00
Recv: echo:  M304 P10.00 I0.02 D305.40

```
i have enabled the eeprom and have tried saving the pid settings there, but it does not remember them.  i have enabled #define FLASH_EEPROM but when i run m500 it tells me the settings are saved, but there is no saved settings on the sd card (just the FIRMWARE.CUR file that says it was successfully flashed.

attached is my configuration.h  am i missing something?  
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/2687065/Configuration.h.txt)

### Steps to Reproduce

1. update Configuration.h to have custom pid values for an MKS Sbase 1.3
2. Compile marlin with platformio (platformio run -e LPC1768
3. copy firmware file to flash card and boot up control board
4. connect with octoprint look at the read out, it has pid values that are not put in configuration.h

**Expected behavior:** for it to use my pid values

**Actual behavior:** it ignores my pid values and uses the ultimaker ones

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

GMagician, 2018-12-17:
Before M500 you have to reset to default by M502

gloomyandy, 2018-12-17:
No the stock values are the ones loaded from eeprom (assuming you have eeprom enabled), as above you need to load the firmware defaults (with m502) and then save those defaults to eeprom.

alexzapatka, 2018-12-17:
thank you !! now i feel silly.  it works exactly as you said.

closed 2018-12-17 by alexzapatka