MarlinFirmware/Marlin #12668: Nodejs with Marlin and its not work

city8208, 2018-12-18:
Hey,I use the Nodejs to send code by MKS GEN(arduino 2560)
But no matter what Gcode I send,I just seem the blue led Shining and Nothing happens.
Could you give some help?

Code:
//////////////////////////////////////////////
  SerialPort = serialport.SerialPort;
    Readline = serialport.parsers.Readline;
    sp = new serialport(settings.serialPort, {
        parser: new Readline("\n"),
        baudrate : settings.baudRate,
    });
    sp.on('open', function (err) {
        sp.on('data', function (data) {
            if(data){
                sp.write(comm + "\n", function(err, result){
                    if (err) {
                        consoleAndPrint(comm+'...'+'Error on write: '+err.message,false);
                        consoleAndPrint('Response received after sending message : ' + data,false);
                        setTimeout(printerCommand(comm), 50);
                    }else{
                        printPosition += 1;
                        consoleAndPrint(comm+'...'+'Success on write',false);
                        consoleAndPrint('Response received after sending message : ' + data,false);
                        if(printPosition < printCommands.length) {
                            printerCommand(GCODE.reconstruct(printCommands[printPosition]));
                        }else{
                            consoleAndPrint('Finish',true);
                        }
                    }
                });
            }
        });
    });
/////////////////////////////////////

boelle, 2019-02-21:
@city8208 problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-10 by shitcreek