MarlinFirmware/Marlin #12670: Full fan causes unrecoverable temperature drop and termal runaway

grandcrossgc, 2018-12-18:
### Description

Turning the fan on full while at temperature causes an unrecoverable temperature drop and thermal runaway. The hotend can heat fine with full fan. I've tried autotuning the PID with fan on and off with no difference in results. I've adjusted the thermal runaway options but I still get the error because the hot end never reheats. It appears not to even try

### Steps to Reproduce

1. Turn the fan on. Bring  the hot end up to the desired temp. No problem.
2. Turn the fan off. Bring the hot end to the desired temp. Turn the fan on. The temperature drops and never recovers, eventually ending in thermal runaway error.

**Expected behavior:** [What you expect to happen]
When the fan turns on, the temperature should drop but recover

**Actual behavior:** [What actually happens]
When the fan turns on, the temperature drops and continues to drop. It never recovers

#### Additional Information

 T:21.95 /0.00 B:23.09 /0.00 @:0 B@:0
 T:21.87 /0.00 B:23.09 /0.00 @:0 B@:0
 T:21.87 /0.00 B:23.05 /0.00 @:0 B@:0
 T:21.87 /0.00 B:22.97 /0.00 @:0 B@:0
 T:21.87 /0.00 B:23.01 /0.00 @:0 B@:0
 T:21.84 /0.00 B:22.97 /0.00 @:0 B@:0
 T:21.87 /0.00 B:22.97 /0.00 @:0 B@:0
 T:21.87 /0.00 B:22.93 /0.00 @:0 B@:0
 T:21.87 /0.00 B:23.12 /0.00 @:0 B@:0
 T:21.87 /0.00 B:23.05 /0.00 @:0 B@:0
 T:21.84 /0.00 B:23.05 /0.00 @:0 B@:0
 T:21.87 /0.00 B:23.01 /0.00 @:0 B@:0
 T:21.87 /0.00 B:22.97 /0.00 @:0 B@:0
 T:21.87 /0.00 B:23.09 /0.00 @:0 B@:0
 T:21.87 /0.00 B:23.09 /0.00 @:0 B@:0
 T:21.87 /0.00 B:23.01 /0.00 @:0 B@:0
 T:21.87 /0.00 B:23.09 /0.00 @:0 B@:0
 T:21.87 /0.00 B:23.05 /0.00 @:0 B@:0
 T:21.87 /0.00 B:22.97 /0.00 @:0 B@:0
 T:21.84 /0.00 B:23.05 /0.00 @:0 B@:0
 T:21.87 /0.00 B:23.01 /0.00 @:0 B@:0
ok <- Set the extruder temp to 210
 T:21.87 /210.00 B:23.05 /0.00 @:0 B@:0
 T:21.87 /210.00 B:23.12 /0.00 @:127 B@:0
 T:22.50 /210.00 B:22.97 /0.00 @:127 B@:0
 T:24.30 /210.00 B:22.97 /0.00 @:127 B@:0
 T:26.36 /210.00 B:23.05 /0.00 @:127 B@:0
 T:28.75 /210.00 B:23.05 /0.00 @:127 B@:0
 T:31.72 /210.00 B:23.05 /0.00 @:127 B@:0
 T:35.06 /210.00 B:23.05 /0.00 @:127 B@:0
 T:38.17 /210.00 B:23.05 /0.00 @:127 B@:0
 T:41.71 /210.00 B:23.05 /0.00 @:127 B@:0
 T:45.16 /210.00 B:23.09 /0.00 @:127 B@:0
 T:48.62 /210.00 B:23.05 /0.00 @:127 B@:0
 T:51.97 /210.00 B:23.09 /0.00 @:127 B@:0
 T:55.39 /210.00 B:23.09 /0.00 @:127 B@:0
 T:59.06 /210.00 B:23.05 /0.00 @:127 B@:0
 T:62.33 /210.00 B:23.05 /0.00 @:127 B@:0
 T:65.62 /210.00 B:23.09 /0.00 @:127 B@:0
 T:68.87 /210.00 B:23.05 /0.00 @:127 B@:0
 T:72.34 /210.00 B:23.12 /0.00 @:127 B@:0
 T:75.51 /210.00 B:23.09 /0.00 @:127 B@:0
 T:78.56 /210.00 B:23.09 /0.00 @:127 B@:0
 T:81.62 /210.00 B:23.12 /0.00 @:127 B@:0
 T:84.61 /210.00 B:23.12 /0.00 @:127 B@:0
 T:87.69 /210.00 B:23.12 /0.00 @:127 B@:0
 T:90.52 /210.00 B:23.09 /0.00 @:127 B@:0
 T:93.38 /210.00 B:23.09 /0.00 @:127 B@:0
 T:96.17 /210.00 B:23.09 /0.00 @:127 B@:0
 T:98.95 /210.00 B:23.12 /0.00 @:127 B@:0
 T:101.84 /210.00 B:23.05 /0.00 @:127 B@:0
 T:104.42 /210.00 B:23.05 /0.00 @:127 B@:0
 T:107.03 /210.00 B:23.12 /0.00 @:127 B@:0
[... 130 lines cut ...]
 T:197.27 /210.00 B:23.12 /0.00 @:127 B@:0
 T:197.27 /210.00 B:23.12 /0.00 @:127 B@:0
 T:197.27 /210.00 B:23.12 /0.00 @:127 B@:0
 T:197.27 /210.00 B:23.12 /0.00 @:127 B@:0
 T:197.24 /210.00 B:23.16 /0.00 @:127 B@:0
 T:197.27 /210.00 B:23.12 /0.00 @:127 B@:0
 T:197.24 /210.00 B:23.12 /0.00 @:127 B@:0
 T:197.22 /210.00 B:23.09 /0.00 @:127 B@:0
 T:197.10 /210.00 B:23.12 /0.00 @:127 B@:0
 T:197.13 /210.00 B:23.09 /0.00 @:127 B@:0
 T:197.22 /210.00 B:23.12 /0.00 @:127 B@:0
 T:197.24 /210.00 B:23.12 /0.00 @:127 B@:0
 T:197.27 /210.00 B:23.12 /0.00 @:127 B@:0
 T:197.27 /210.00 B:23.09 /0.00 @:127 B@:0
 T:197.27 /210.00 B:23.12 /0.00 @:127 B@:0
 T:197.24 /210.00 B:23.12 /0.00 @:127 B@:0
 T:197.24 /210.00 B:23.12 /0.00 @:127 B@:0
 T:197.24 /210.00 B:23.05 /0.00 @:127 B@:0
 T:197.27 /210.00 B:23.12 /0.00 @:127 B@:0
 T:197.27 /210.00 B:23.12 /0.00 @:127 B@:0
 T:197.24 /210.00 B:23.12 /0.00 @:127 B@:0
 T:197.27 /210.00 B:23.09 /0.00 @:127 B@:0
 T:197.27 /210.00 B:23.12 /0.00 @:127 B@:0
Error:Thermal Runaway, system stopped! Heater_ID: 0
Error:Printer halted. kill() called!

grandcrossgc, 2018-12-18:
[marlin_config.zip](https://github.com/MarlinFirmware/Marlin/files/2688908/marlin_config.zip)

RudyBenner, 2018-12-18:
#define FAN_KICKSTART_TIME 100
#define FAN_MIN_PWM 50
#define FAN_MAX_PWM 160

configuration_adv.h

swilkens, 2018-12-18:
Side idea: Is the fan potentially causing too much airflow on the heater block? The airflow should be directed at the nozzle tip, preferably slightly below.

This because you said you tuned with the fan on as well, leading me to believe the airflow overpowers the heating cartridge.

comps, 2018-12-18:
I'm seeing the same thing on about 8 month old Marlin (pre-1.1.9) but I'm unable to flash new versions for some reason, so I can't help much.

I have the e3d silicone sock, so the fan isn't really blowing the heat away and the board+PSU have plenty more power to spare and I did all the PID tuning with fan enabled just to be safe.

The problem (again, on my old Marlin) is that the PID algorithm seems broken somehow. If I tell it to heat to ie. 280C, it heats rapidly to 270C until PID kicks in at which point it falls to ~255C before kicking up again and triggering thermal runaway around 265C.

The thing is - this happens with temperatures as low as 200C too, so it is the PID not pumping in enough duty cycle to properly keep temperature, despite my ~40 attempts at tuning. Yes, I have the default `BANG_MAX` == `PID_MAX` == 255.

See if you can reproduce the issue with the fan physically pointing elsewhere, away from the heater block.

grandcrossgc, 2018-12-18:
> #define FAN_KICKSTART_TIME 100
> #define FAN_MIN_PWM 50
> #define FAN_MAX_PWM 160
> 
> configuration_adv.h

As a work around, I'll take it. I don't consider this a solution though.

The heater block can heat fine with the fan on. It should be able to reheat. It did with previous firmware versions.

grandcrossgc, 2018-12-18:
> Side idea: Is the fan potentially causing too much airflow on the heater block? The airflow should be directed at the nozzle tip, preferably slightly below.

It probably is, although it's in a fixed position so I can't really move it without major mods. I plan to do that, but the core problem is still there. The heater should be able to reheat.

RudyBenner, 2018-12-18:
Also dial back the heater, its a bit 'peppy'. Doing so makes a better PID.

#define BANG_MAX 200

Do another PID after the change.

AnHardt, 2018-12-18:
```
T:197.27 /210.00 B:23.09 /0.00 @:127 B@:0
T:197.27 /210.00 B:23.12 /0.00 @:127 B@:0
                                 ^^^
Error:Thermal Runaway, system stopped! Heater_ID: 0
Error:Printer halted. kill() called!
```
PID is working and orders the PWM to give all in.
So something, if not the airflow, is disturbing the PWM at the heater pin.

Try soft pwm for the fan. Check what timer the fan-pwm is using if analogWrite() is used.

viper93458, 2018-12-19:
Just a quick note to say that I have also seen this lately with 2.0 bug fix but I have not made any attempts to diagnose or further troubleshoot, so I can’t say it’s a bug, but something seems to have changed at some point. I have dialed back my fan speeds to address it for now.

William

razerraz, 2018-12-30:
I'm facing similar issue on my anet A8, seems since my last marlin upgrade (marlin bugfix 1.1.x, released in something like may 2018 as my file system point out) : my cooling fan turns on on layer 2 then the temp drop dramatically and never goes up really, standing in a warning low temp value.
 
I've suspected the resistor first, comparing with new one with an ohmmeter, then checked the electronic part (trying a mosfet module).
Then I checked software side, making calibration, with/without fan, try my own values.

What I see tells me there is something wrong software wise, since everything is fine on idle (setting the end near the bed, fixing a temp + bed value, turning on/off the fan, in the same condition than when printing), and during printing problem appears.
I suspect that the controller is too much busy reading and executing gcode lines and forgot to take real care of the pid computing.

Will try upgrade bugfix 2.0.0 and tell

razerraz, 2019-01-01:
Response to myself
I've found the cause of my issue : the temperature probe was not correctly inserted in the hotend, and was moving during printing.
Nothing Marlin related though

grandcrossgc, 2019-01-07:
An update on this... it may be hardware issues after all. I tried to print some PETG the other day and the temperature stuck at 226, even though I'd printed a lot of ABS at 230 earlier this year. I may have a worn out heater. I've a replacement on order and I'll retest everything once it's installed.

athulsnair, 2019-02-11:
Same happens with my printer too. But it runs on Marlin 1.1.6. I have donw PID tuning with fans ON. When not printing I can heat the hot end to 205 with fans ON. Also as test I tried to heat the coil to 205 degress with fans OFF, then turning it ON to see the effect.Temperature falls a bit, but recovers soon. But during print, when Fan turn ON temps goes below 200. My power supply is rated for 20 A, also in my board RAMPS 1.4 I can see the heating LEDs are ON

Already increased 
```
#define THERMAL_PROTECTION_PERIOD 120 // Seconds
#define THERMAL_PROTECTION_HYSTERESIS 3
```

AnHardt, 2019-02-11:
Is it failing with "Heating failed" or "Thermal runaway"?

The 'printing' heater needs a different amount of energy.
\+ Nozzle moves relative to the sorounding air - more cooling
\+ Nozzle has to melt filament
\+- Part/bed below the nozzle alters/reflects airflow 

**If** the heater power is sufficient at all, you could try a bit more 'agressive' PID parameters
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/temperature.cpp#L469-L475
on cost of a bit more overshoot.

A bit of 'not electrically conductive' but 'thermal conductive' material (Thermal Compound) between heater cartridge/sensor and heater block may help wonders.

InsanityAutomation, 2019-02-11:
If you keep the fan off until temps are stable, you can use adaptive fan slowing in current branches of marlin 2 in order to prevent the fan from cooling too much. Works best with the hysteresis just above it at 10c or so. This sounds like exactly what it was designed for.

razerraz, 2019-02-12:
Other option is to get a silicone cover for the hotend available on most suppliers. Cheap and efficient, at least for me...
And by the way I'm still surprised about all the things this firmware can do...

AnHardt, 2019-02-12:
While we are at this track we should mention how helpful a proper fan-duct is with this.

grandcrossgc, 2019-02-14:
Finally got around to changing my heater. Some observations, but the TL;DR is this is a symptom of a still usable but diminished heater.

Observations:
1. The effect is pronounced when the fan can reflect off of a surface. This includes the bed and the part itself. I was able to run the fan fine when sufficiently spaced from the bed, but when the distance became close I'd run into the thermal runaway.
2. Extruding plastic reduces this tolerance. I know... duh. But it meant I wasn't able to test the distance from the bed and then reliably extrude a model. Even with the air bouncing off of the model I would get thermal runaway.
3. Fan ducts work, but are limited due to the air bouncing off of the model. Not sure about the silicon sock.
4. I had noticed over the years that my temp was always set a bit higher than recommended for the materials I was using. I think this is why. As the temperature fell, it was still warm enough to extrude.
5. Reducing the fan speed resulted in insufficient cooling of the part. Any bridging was just plain unacceptable. This was not a useful workaround.

Overall, I'm not sold on this feature. I've heard anecdotal evidence of other people experiencing similar problems, possibly for the same reasons. My next step prior to replacing the heater was to disable this feature altogether so I could back to my previously acceptable prints. I realize it's supposed to be a safety feature, but I think most of the failure modes it would catch are already handled with min and max temp settings.

It's enabled, for now. Let's see what happens in a few months when the new heater degrades.  I want to leave this open for a little while so people can add their comments and opinions.

boelle, 2019-02-21:
@grandcrossgc problem fixed?

grandcrossgc, 2019-02-22:
I guess so. Closing this with mixed feelings though.

closed 2019-02-22 by grandcrossgc

boelle, 2019-02-22:
you can open a new one in future if problem is still there after trying bugfix 2.0

and check that your fan is not cooling down the heat block, i have had that issue where fan is cooling the heatblock and not the areau just under heat block