MarlinFirmware/Marlin #12672: missing nozzle icon?

lulu3d, 2018-12-18:
Hi,

I just installed the latest bugfix 2.0, and I have no nozzle icon on my lcd display (general reprap lcd). The former bugfixes 2.0 I used, I never encountered this issue. 
Is this a bug, or am I missing something? 
I installed this build 2 times and both give me only temperature display, but no nozzle icon (neither when heating up). (of course this is not at all a deal braker, just wondering ;-))

Glod76, 2018-12-18:
post your configs for help. zip configuration.h and configuration_adv.h

lulu3d, 2018-12-19:
[Marlin Configs.zip](https://github.com/MarlinFirmware/Marlin/files/2694041/Marlin.Configs.zip)

OK here they are.

lulu3d, 2018-12-20:
[image]
Just the screen capture of the issue ;-)

rmoravcik, 2018-12-20:
@lulu3d Your Configuration_adv.h is outdated. One of the missing sections is causing the missing the nozzle icon:
```
  /**
   * Status (Info) Screen customizations
   * These options may affect code size and screen render time.
   * Custom status screens can forcibly override these settings.
   */
  //#define STATUS_COMBINE_HEATERS    // Use combined heater images instead of separate ones
  //#define STATUS_HOTEND_NUMBERLESS  // Use plain hotend icons instead of numbered ones (with 2+ hotends)
  #define STATUS_HOTEND_INVERTED      // Show solid nozzle bitmaps when heating (Requires STATUS_HOTEND_ANIM)
  #define STATUS_HOTEND_ANIM          // Use a second bitmap to indicate hotend heating
  #define STATUS_BED_ANIM             // Use a second bitmap to indicate bed heating
  //#define STATUS_ALT_BED_BITMAP     // Use the alternative bed bitmap
  //#define STATUS_ALT_FAN_BITMAP     // Use the alternate fan bitmap
  //#define STATUS_FAN_FRAMES 3       // :[0,1,2,3,4] Number of fan animation frames
  //#define STATUS_HEAT_PERCENT       // Show heating in a progress bar
```

lulu3d, 2018-12-20:
OK, thanks @rmoravcik . 

I must have messed this file up during upgrades?! 
I'm so sorry to have bothered you with this, but I'm glad you could find me the issue. Top, for helping me out :-)

closed 2018-12-20 by lulu3d