MarlinFirmware/Marlin #12673: bug with Marlin 2.0.x alpha snapshot

stefzouille, 2018-12-18:
hello, I just opened the marlin.ino with arduino 1.8.8 and 1.8.5 and without changing anything by selecting the correct com com puts me this message, I'm a beginner and my English and not terrible, version 1.1.9 I manage to teleport without any worries .. if you have an idea I am interested. cordially 👍 

error : 

C:\Users\conta\AppData\Local\Temp\cc5fYtY8.ltrans0.ltrans.o: In function `main':

C:\Users\conta\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.23\cores\arduino/main.cpp:43: undefined reference to `setup'

C:\Users\conta\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.23\cores\arduino/main.cpp:46: undefined reference to `loop'

collect2.exe: error: ld returned 1 exit status

exit status 1
Erreur de compilation pour la carte Arduino/Genuino Mega or Mega 2560

RudyBenner, 2018-12-19:
[image]

RudyBenner, 2018-12-19:
Also select your com port

stefzouille, 2018-12-19:
Hello,
really sorry to be poorly explained, but the type of card and the port com and the atmega328 is beautiful and well selected! I do the manipulation with the marlin 1.1.9 and it works, even by changing any parameter with the 2.0 just open the .ino with arduino 1.8.8 and check right away by selecting the mega2560 com com it does not work j always have this error message, I imported the zip in the crosuis but it does not change, at home it does not work!

stefzouille, 2018-12-19:
I see this error message with the 2560, but I do not understand why he puts me this error, while with 1.1.9 it works well?!? I even configure the 2.0 with my parameters for my 1.4 ramps but I still have this error, I start immediately at 5 minute intervals with the 1.1.9 and it works, I close and opens immediately 2.0 and that puts an error.?!?

stefzouille, 2018-12-19:
with a friend he made me notice that I had separated marlin.ino + configuration.h and adv.h from the rest of the file, the normal shot that it did not walk ..

closed 2018-12-19 by stefzouille