MarlinFirmware/Marlin #12674: [bugfix-2.0.x] Bltouch error

tifrei, 2018-12-18:
### Description

connected my Ramps that i used before with and Arduino Mega to my newly Re-ARm with Marlin 2.0. Once it powers up the Bltouch just blinks red. Cant be a wirign issue because when i put the Ramps back on the Mega it works just like a charme. 

### Steps to Reproduce

1. Connect Bltouch to Ramps
2. Put ramps on the Re-Arm
3. start it up and see the bltouch blinking red

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

kAdonis, 2018-12-18:
It **can be** a wiring issue because the Re-Arm provides only 3.3V to the RAMPS servo port (via Jumper)
The Mega provides 5V, 
You need to connect the yellow BLTouch cable to 5V https://www.antclabs.com/wiring32
or remove the jumper on RAMPS and connect 5V to the RAMPS servo port
https://github.com/MarlinFirmware/Marlin/wiki/Servo-Power

tifrei, 2018-12-19:
Hi Thanks for the help @all. I guess @kAdonis is right. Ill try it later today and let you know. thanks.

kAdonis, 2018-12-19:
@FanDjango 
As I understand, cutting the trace on the BLTouch just changes the output ( high) level from 5V to 3.3V.
which is important for 32bit boards that are not 5V tolerant. The Re-Arm is 5V tolerant (except pins 1_30 and 1_31)

FanDjango, 2018-12-19:
@kAdonis 
You are right and I was to quick off the mark. As you say, it seems the BLTouch needs a 5V supply. And it also seems it can live with a 5V or 3.3V servo signal (yellow wire).

Bob-the-Kuhn, 2018-12-20:
I was getting flakey performance on my Due when I had the BLTouch powered by 3.3V.  Going to a 5V supply made it a (mostly) happy camper.

My BLTouch performs nicely but it is usually blinking red.  This is the "slow red blink" which can't be cleared by using the reset command.  It means that the unit isn't happy with Vcc.   As best I can tell Vcc is sampled at power up and the slow blink is started if Vcc isn't in the desired range.

The fault blink is faster.  If you force a fault condition then it'll switch to the faster blink.  Issuing the reset command puts it back into the slow blink.

The slow blink is documented in Antclab's [wiring app note](https://www.antclabs.com/wiring).