MarlinFirmware/Marlin #12675: G29 P1 Probing in 1.1.9 broken (worked fine in previous Marlin)

Freakadude, 2018-12-18:
I was using Marlin 1.1.7 before and upgraded recently to 1.1.9. Everything works fine on the printer as expected appart from the UBL command G29 P1. 

When i do a G29 P1 under Marlin 1.1.7 the UBL mesh run completes normally. I have set multiple probing to 3 so each point is probed 3 times. 1st is fast 2nd and 3rd are slow. 

If i initiate G29 P1 command on 1.1.9 (with same core settings as in my 1.1.7 config) then after the first fast probe basically the machine comes to a stop. After looking i noticed it does not actually stop but the subsequent probe movements are slowed to literally crawling speeds. The motors give a weird noise and i can see the z axis moving ever so slightly. But basically what should be a second probe in 2-3 seconds turns into a 10-15 second or longer situation. Home commands or standard move commands all complete normally only the G29 P1 command seems to be broken in this way for some reason.

[https://youtu.be/yLHkeObAQOQ](url) You can see the issue from 0:34 in the video.

Appriciate the help.

Roxy-3D, 2018-12-18:
I installed 1.1.9 on my printer about 4 weeks ago and did a full calibration cycle to bring it up.   I did not see any issues with the G29 P1.     Can you attach your Configuration.h files to the thread so we can look closer at what you are doing?

And then a few questions:

- Is that a 'Real' BL-Touch probe?     Can we do an M48 at the home location and verify that we can trust the BL-Touch probe results? 

- Also, can you home the machine, and then very carefully...  very slowly...   lower the nozzle until it just touches the bed.   What is the Z Height value on the LCD Display when the nozzle is just touching the bed?

Freakadude, 2018-12-18:
It is a real BL touch and not a clone.
Sorry but your second question doesen't make any sense in context of my problem.

To reiterate: Every normal function on the printer (movements, homing, heating, extruding, etc) work fine just as they did in the previous Marlin setup i had. 

Configuration.h file is basically the same as i had in 1.1.7 with the exception of the newly added things in 1.1.9 of course but basic machine settings (speed, jerk, end stops, bed leveling etc) all remain the same.

Roxy-3D, 2018-12-18:
> Sorry but your second question doesen't make any sense in context of my problem. What does the Z offset height have to do with anything?

Most problems with G29 P1 ultimately come back to somebody not getting the machine fully calibrated and able to print a small object in the center of the bed without bed leveling active.   The probing functions tell the nozzle to go 10 mm (or so) below the surface of the bed.  If Z_OFFSET_FROM_EXTRUDER is incorrectly set, it is very possible the probing won't work.

> Configuration.h file is basically the same as i had in 1.1.7 with the exception of the newly added things in 1.1.9 

It would be worth while to do a visual diff between your configuration.h files and the 1.1.9 default file.  Some options have changed names.   And if an important option changed names and you did not catch it, it would be turned off.

Freakadude, 2018-12-19:
Z offset is set by me every once in a while to adjust for correct print height but its not maintained in the config file but in the printer eeprom. Thats not the issue here.

I did not just copy and paste settings from 1.1.7 into 1.1.9 file but went line by line and identified same settings between 1.1.7 and 1.1.9 and updated accordingly. 

Again for me the main point is (which is visible in the video) that homing works as intended, first fast probe on the first point works as intended but things go to hell after first point second (first slow) probe. If i had to guess something with regards to the slow probe speed and the multiple probe setting is causing this issue on a first glance. I will check today by disabling multiple probing to see if that way the G29 P1 works correctly or not.

Freakadude, 2018-12-19:
Well i just tryed disabling Multiple Probing and as a result now when i issue the G29 P1 command the probe moves to the first point position and then just stops without moving down of continuing to any further point. So something is deffinately buggered with G29 here.

BTW i am using a Creality CR-10S if that makes any sense to mention at this point.

Freakadude, 2018-12-19:
Just like to ad that i have just tryed the bugfix 1.1.9 release but have the same result.

For further analysis i have attached my working Marlin 1.1.7 config files and the Marlin 1.1.9 config files where this G29 issue occures.

[https://www.dropbox.com/s/4k4iehd30b7owq4/Configuration_%28v1.1.7bugfix%20working%20G29%29.h?dl=0](url) Working 1.1.7. config.h file

[https://www.dropbox.com/s/8dvr2piegk73c54/Configuration_adv_%28v1.1.7bugfix%20working%20G29%29.h?dl=0](url) Working 1.1.7. config_adv file.

[https://www.dropbox.com/s/69giuc3mog2vkls/Configuration_%28v1.1.9%20NOT%20working%20G29%29.h?dl=0](url) NOT working G29 with 1.1.9 config file.

[https://www.dropbox.com/s/6ea76g7mu5nw91o/Configuration_adv_%28v1.1.9%20NOT%20working%20G29%29.h?dl=0](url) NOT working G29 with 1.1.9 config_adv file.

Roxy-3D, 2018-12-19:
> You should rebuild the firmware with #define DEBUG_LEVELING_FEATURE turned on. And then do a M111 S255 and repeat the G29 P1. If you post all the (very verbose) output of that, we should be able to figure out what is not working correctly.

This will give us a lot of information about what is going on in the firmware.

Freakadude, 2018-12-19:
Tryed to rebuild firmware as suggested by enabling the mentioned feature in configuration.h and i get the following compile error message: exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

Freakadude, 2018-12-19:
Okay i rebuilt the firmware with the debug feature turned on and just did a G29 P1 after running command M111 S255. The weird thing is now the G29 P1 suddenly works (altough somewhat slower then normally which may be due to the intensive logging in the background) and goes through the motions and points as it should HOWEVER i hear that for some reason the part cooling fan is triggered on and off ever so slightly between probe moves which is also a very weird byproduct that normally does not happen when i do a G29 P1 probe. Additionally after the probing of all points is completed I now for some reason get a insert filament prompt on my printer which is also completly random as filament IS inserted and nothing has been changed or touched in that regard.

So as a next step i did a printer reset and went ahead and issued a G29 P1 command WITHOUT issuing a prior M111 S255 command and the probing is done now also. The strange sideeffect of the part cooling fan being triggered randomly with a low output and then being shut down is still there though which i do not understand either as again this is not something that is happening on firmware before 1.1.9.

The output during M111 S255 seems normal enough to me but here is a bit of it for you to see if that helps. I could only copy part of the end of the process as the start of the log was cut off after a while in my console unfortunately. However while i watched i saw no change in the outputs compared to what is visible in the end of the log.

[https://www.dropbox.com/s/amc860t28ohqwmn/G29%20P1%20log%20%28end%20of%20run%20%2C%20no%20start%29.txt?dl=0](url) G29 P1 log (partial w/o beginning and just the end visible)

As a last test i reflashed the 1.1.9 firmware with only one adjustment: I again disabled the #define DEBUG_LEVELING_FEATURE to see if this one change really has an affect on G29 P1 being executed on the printer or not. And weirdly now the G29 P1 works for a couple of probe points normally but after about 3-4 it again stops with the weird stepper sound and the Z axis basically not moving...

I am not a programming expert but this really dosen't make any logical sense to me. 

Maybe also worth to mention that ever since 1.1.9 when i compile the firmware I always get a low memory warning at the end which i never got with prior version of Marlin firmware. Basically it sais that Global variables use 91% of dynamic memory and only 709 bytes remain free. Maybe this has some sort of impact on the G29 P1 command?? But even if that is the case then why does the same G29 P1 command run through successfully if the DEBUG_LEVELING_FEATURE is enabled and not when its disabled regardless if the M111 S255 command is run beforehand or not?

So basically two issues remain: 1) The G29 P1 command only seems to run through correctly IF DEBUG_LEVELING_FEATURE is enabled. It runs through regardless if M111 S255 command was issued beforehand or not. If this feature is not enabled then G29 P1 either dosen't even get through 1 probe point or for some reason it gets 3-4 points done and then goes back to stopping/slowing to a usless crawl. 2) In the scenario when G29 P1 works as mentioned in point 1) the part cooling fan is randomly triggered on and off with random small performance settings. This is also weird.

Related to issue 2) i also see on the LCD display of the printer that the fan shows random weird %s. It shows all sorts of numbers (also negative %s) while doing the probing. Also feedrate on the LCD shows a bunch of weird random values as well ranging from 999% to other values. Really don't think this is also normal behavior.

Roxy-3D, 2018-12-19:
> Basically it says that Global variables use 91% of dynamic memory and only 709 bytes remain free. 

This number is much too low.   We need to free up some memory.   You need to turn off some options that you don't really need.

Freakadude, 2018-12-19:
Ok i am open for suggestions. What features should i turn off to increase memory? BTW don't want to beat a dead horse but prior to 1.1.9 i never had memory issue warnings. Just saying that don't know what eats up so much memory since 1.1.9 but if there are features that take up lot which i don't even use i gladly will disable them if you can point me in the right direction. i just don't want to lose functionality compared to my 1.1.7 setup as then i don't see the point in upgrading to 1.1.9.

Yes i did do M502 and M500.

Roxy-3D, 2018-12-19:
Do you need to print from an SD-Memory card?  If not...  That might free up enough memory that everything else can be left alone.

Freakadude, 2018-12-19:
I don't need SD card as i print via Repetier Server through a RPI3. Which lines should i disable for this? #define SDSUPPORT in config.h enough to disable?

Roxy-3D, 2018-12-19:
For future reference...   It is 'safe' to have 1800 or 1900 bytes of memory free.   If you get much below that, you start seeing random problems.

Freakadude, 2018-12-19:
It seems after freeing up memory by disabling the SD Card support the issue seems to be resolved. G29 P1 is doing its thing now under 1.1.9 and also the weird side effect of the fan and LCD display is gone now too. 

Just an idea based on this investigation: I often contemplated this memory warning however inside of the config.h file or the config_adv.h file there are not really many mentions (appart from a few lines where its explicitly mentioned disabling them can free up memory) what functions take up how much chunk of the memory. After the low memory warning i always contemplated what i could disable to free up memory but had no real indication based on the comments as to where i should focus. Maybe there could either be a wiki or perhaps more mentions of memory heavy functions inside the config files so in cases like this a user runs into memory warnings they can contemplate what features they might want to sacrifice in order to get a still good running system. 

At any rate thanks for the support here. I appriciate it.

Roxy-3D, 2018-12-19:
> Maybe there could either be a wiki or perhaps more mentions of memory heavy functions inside the config files so in cases like this a user runs into memory warnings they can contemplate what features they might want to sacrifice in order to get a still good running system.

It would be good if somebody could compile the firmware with each option turned on and off and report the code usage.   If we had a table like that, it would help people decide what options they value enough to warrant inclusion in their build of the firmware.   Our project is entirely volunteer based, so somebody needs to both see the value and have time and energy to accomplish the task.   (And most of our volunteers don't like documentation, so that raises the bar for something like this.)

With that said...   32-bit boards are getting better supported and much cheaper.   Soon (1 or 2 years???) the 8 bit boards will not be used much.   And even with today's 32-bit boards, neither RAM or ROM space is problem.    So it is very possible, this becomes a non-issue in a few years.

boelle, 2019-02-21:
@Freakadude problem fixed?

Freakadude, 2019-02-22:
> @Freakadude problem fixed?

Yes but i alrady said as much in my reply on Dec 19 above.

closed 2019-02-22 by Freakadude

thinkyhead, 2019-02-23:
Thanks, @Freakadude — There's a fishing expedition on lately to get some issues closed.

Roxy-3D, 2019-02-23:
> Thanks, @Freakadude — There's a fishing expedition on lately to get some issues closed.

Well, this fishing expedition is causing a lot of people trouble.   

Every "Is this fixed?   Can this be closed?" generates an email.    And it also makes old issues and pull requests change their status to 'Recently Updated' which means people can not use that as a filter to see what items are 'Hot'.

boelle, 2019-02-23:
well, the alternative is that we do nothing and it will all end up like a gian croc of poop

thinkyhead, 2019-02-25:
It has been an interesting experiment. I have my email notifications set to minimum, so this round of cleanup didn't flood my inbox. I was planning to spend this past week on cleaning up issues anyway, and there are a lot I would have just closed with a comment. I'm not sure if my efforts would have generated a similar flood.

Anyway, I was able to follow through on reviewing PRs throughout the week, while also picking through issues in between Travis CI tests. I started out with nearly 700 unread notifications. And now I still have over 600. So, stirring up the unread issue queue for me didn't help with my original plan of getting the unreads down under 100. But, I can follow through on that this week.

I was able to skip over the issues that were last-commented on by Bo, since these stand out in the list and there's a "read" checkmark in the list. Without that helpful facility my unread count would be heading toward 1000.

So, thanks Bo for taking the time to pore through and get many issues actually closed. I'm going to pick up with my plan from last week and see if I can close another 100 or 200 this week, working backward from oldest issue to newest. As I mentioned before, I'll leave unfulfilled [FR] and bug report issues open, tagged as such.

As I go through unresolved issues, I always try to solve them right away if possible. Very involved or complicated issues get passed over until later. So the aim is to have all the open issues be the harder ones, and for all the easy issues and major bugs resolved ASAP.