MarlinFirmware/Marlin #12677: Layershift/missing steps with MKS SBASE

palmhoej, 2018-12-18:
[image]

When running Marlin 2.0 (downloadet 3dec) on a MKS SBASE i see layershifts/missing steps on my print (see pictures) from my Tronxy X5s (CoreXY printer).

In this video you can clearly hear it losing steps (around 10 seconds and 25 seconds in the video):
https://www.youtube.com/watch?v=6rsMBklIbGw

As also seen from video this results in skewed prints.

What i have checked:
- All mechanical are perfect, this is not induced from mechanical resistance (or the nozzle hitting the object). I know that the picture shows to identical prints with layershifts in the exact same spot, which seems to indicate a z-axis fault. So i moved the z-axis 20mm, but the layershifts came at exactly the same spot.
- different models and slicer used, same problem, but different places the layer shifts are happening (supporting that it is not mechanical!).
- Checked with stepperdriver on the MKS Board and external stepperdriver (Leadshine DM422 current install seen on the video on X & Y motors). Same behaviour, although more pronounced with external stepperdriver.
- Tronxy s5x config profile have been used, but acceleration, speed and jerk values have been dialed down by roughly half to see if this helped, it did not).
- Pulse width set to 20 us for external steppers (minimum of 7.5 us required). Checked with scope (time of picture it was set to 30 us!):
[image]

- All changed i have made in config files have been marked with KPS 
[Configuration_Files.zip](https://github.com/MarlinFirmware/Marlin/files/2692686/Configuration_Files.zip)

mgrl, 2018-12-19:
This was reported already. It seems there is no final solution yet. Find more information here (#11577) and there (#12403).

Roxy-3D, 2018-12-19:
I suggest you re-load your firmware with a snap shot of the code from Sept. 1st 2018.   Most likely, your layer shift problem will disappear.

Knives79, 2018-12-21:
I have the same issue, how can I get a 1st Sept snap shot to test? Thanks.

hectori4502, 2018-12-21:
checking your configs.
If you use the on board DRV8825 
you need to uncomment
#define X_DRIVER_TYPE  DRV8825
#define Y_DRIVER_TYPE  DRV8825
#define Z_DRIVER_TYPE  DRV8825

#define DIGIPOT_I2C
change 8 to 5 #define DIGIPOT_I2C_NUM_CHANNELS 5 //MKS SBASE: 5
#define DIGIPOT_I2C_MOTOR_CURRENTS { 1.0, 1.0, 1.0, 1.0, 1.0 }  //MKS SBASE  I use xyz 1.2 E 1.1

#define MINIMUM_STEPPER_DIR_DELAY 650
#define MINIMUM_STEPPER_PULSE 20 //KPS uncommented //I change the 2 by 4 and solve the problem
#define MAXIMUM_STEPPER_RATE 250000

And if you set the new backlash, all things will go better.

MoellerDi, 2018-12-29:
I used to have similar issues on my Tronxy X5S.
`#define MINIMUM_STEPPER_PULSE 4` in addition to setting XYZE `DRIVER_TYPE DRV8825` resolved it for me.

boelle, 2019-02-21:
@palmhoej problem fixed?

closed 2019-03-03 by thinkyhead

rally25rs, 2020-02-17:
Same issue on SKR 1.3 w/ TMC 2209 drivers. Some combination of `MINIMUM_STEPPER_PRE_DIR_DELAY`, `MINIMUM_STEPPER_POST_DIR_DELAY`, `MINIMUM_STEPPER_PULSE` fixed it for me too. 👍

J-PN, 2020-03-02:
I got interesting layer shifts with 2.01.

In CoreXY left motor had pushed the head 18mm towards right front corner. 
15 layers later left motor pulled the layer back 20mm towards left back corner.

Cura (4.5) simulation looks OK, hopefully 2.0.4.4 works.

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4276966/Configuration_adv.zip)
[image]