MarlinFirmware/Marlin #12680: [bugfix-2.0.x] Question MKS Robin TFT 320x240

jmz52, 2018-12-19:
I've pieced together some code to display generic 12864 screen on MKS Robin's 320x240 TFT.
https://ibb.co/R9rppgS

According to https://github.com/MarlinFirmware/Marlin/issues/12539#issuecomment-445914714 there might be 6 different controllers used on MKS Robin TFT and there is no way of knowing controller type used in particular display.

There are several possible ways to deal with this problem and I would like to hear your thought on them
- Skip controller initialization. Built-in bootloader performs screen initialization before starting Marlin. Drawing functions (0x2A, 0x2B and 0x2C) are controller-independent, or at least I am yet to find controller that does not support them. Downside of this solution is a must-have requirement for original bootloader, which should be ok for normal users.
- Make user responsible for knowing board's hardware - make several configuration options one for each type of controller.
- Make universal driver ('device' function) that check screen id (Reg04) and use appropriate initialization sequence. This approach will require new U8G_DEV_MSG call to read controller ID in 'communication' function and report it to 'device' function.
- Move initialization and other device-specific code to 'communication' function and create universal 'communication' function.
- Combine options 1 and 2. Create generic driver for Robin and bunch of controller-specific drivers for advanced users or other board.

@thinkyhead, @xC0000005 I would appreciate your feedback

xC0000005, 2018-12-19:
> According to #12539 (comment) <https://github.com/MarlinFirmware/Marlin/issues/12539#issuecomment-445914714> there might be 6 different controllers used on MKS Robin TFT and there is no way of knowing controller type used in particular display.

The display versions I gave you came from the bootloader. 

I’m facing the same sort of question in regards to the lerdge boards (which only support two different controllers, with exactly one change between them). The problem with letting the bootloader init is that at least on the lerdge boards, it isn’t fully initialized unless there’s a firmware upgrade. I can manually muck with registers and draw on the Lerdge screen so long as I’ve flashed new firmware on. I wouldn’t be surprised if the MKS-Robin does something similar where it loads in two different modes, so I’m not sure you can rely on it to do the work.

When it comes to universal display drivers, the Robin appears to support some screens hooked up via SPI and (most) via FMC, so that increases the load. 

My recommendation is to make a class that works for your display and extend as needed. I was seriously thinking of a graphics class that has an array of “init sequence” commands. If you have one for “delay long” then it would just be streaming a set of reg/cmd/delays.  I agree that once you’re up and running, the 2A/2B/2C combination seems universal, so adding a new screen would be an issue of saying “These commands are issued via FMC” or “via SPI” and “here’s the blob of reg/cmd/delay to send.”

jmz52, 2018-12-20:
MKS Robin performs full screen initialization and writes "Booting" and "Completed" even if there was no firmware upgrade. Or at least version 2.4 does.
I see you point - solve one problem at a time, so I will start with ST7789V I have.

jmz52, 2018-12-21:
@xC0000005, 
I've uploaded my code for 12864 emulation on 320x240 TFT screen.
The FMSC initialization is for F1 and can not be reused for F4 without modification. At the very least AF12 should be specified for FSMC pins in F4.
Rest of the code should be more or less universal.

Please note that I've replaced standard dev_fn to com_fn call 
`u8g_InitCom(u8g, dev, U8G_SPI_CLK_CYCLE_NONE);`
with
`dev->com_fn(u8g, U8G_COM_MSG_INIT, U8G_SPI_CLK_CYCLE_NONE, &lcd_id);`
to get controller ID to dev_fn and use this id to select appropriate init sequence.

boelle, 2019-02-21:
@jmz52 problem solved?

closed 2019-02-21 by jmz52