MarlinFirmware/Marlin #12681: BL-Touch sometimes deployed pin while probing.

BTopbas, 2018-12-19:
I have BLTouch Clone (Trianglelab store in aliexpress). I just updated my CR-10 with marlin 2.0 bugfix (also 1.1.9 bugfix) and problem happening.

Here is video: https://www.youtube.com/watch?v=14cpm1zyswc

If you watch video sometimes pin dropping. But g29 works for me. I don't know probing true or wrong. And i have another problem too. When i control machine with control box. When i push menu button on lcd sometimes bltouch deploy and strow quickly. Not a every press. surfing lcd menu and sometimes deploy when i press whell button. I used same bltouch without problem with 1.1.7. I changed only firmware.

M43 S test output:

> Send: M43 S
> Recv: Servo probe test
> Recv: .  using index:  0
> Recv: .  deploy angle: 10
> Recv: .  stow angle:   90
> Recv: . probe uses Z_MIN pin: 20
> Recv: . uses Z_MIN_ENDSTOP_INVERTING (ignores Z_MIN_PROBE_ENDSTOP_INVERTING)
> Recv: . Z_MIN_ENDSTOP_INVERTING: false
> Recv: . deploy & stow 4 times
> Recv: please trigger probe
> Recv: BLTouch compatible probe detected - pulse width (+/- 4mS): 8
> Recv: ok

kipoopik, 2018-12-20:
you might want to try adjusting the small screw, failing that try this https://www.youtube.com/watch?v=eDfGZ4bjxiM or if yours has a magnet trim a tiny bit of the black sleeving from the magnet end

BTopbas, 2018-12-20:
> you might want to try adjusting the small screw, failing that try this https://www.youtube.com/watch?v=eDfGZ4bjxiM or if yours has a magnet trim a tiny bit of the black sleeving from the magnet end

I watched this video before opening bug. And i tried re magnetize small screw. Also i added small magnet on the screw. Not a solution. Magnetic is powerfull. When i pull out pin magnet quickly suck pin. I think this is not a magnet or screw issue. 

Here is another open issue with same problem: https://github.com/MarlinFirmware/Marlin/issues/12304

stefan85, 2018-12-21:
I have the same issue with an LPC1769 processor. It almost seems that the processor is to fast or something The fix you described works perfectly. I hope this can be implemented in Marlin in future builds.

boelle, 2019-02-21:
@BTopbas problem solved?

closed 2019-03-03 by thinkyhead