MarlinFirmware/Marlin #12692: Use of Y splitter and 1 nozzle, and so 1 part fan, Marlin difines 2 fans!!

lulu3d, 2018-12-20:
OK, I don't know if this is a bug, but it certainly bugged me, is the fact that when I configured my Marlin 2.0 to be used with an Y-splitter (so 1 nozzle and 1 part fan) I standard get 2 part fans definitions. So when I change tool, the part fan is not working for the second tool, as I don't have it! 
I use a MKS base 1.5 as board definition.

I finally resolved the situation in the pins_RAMPS.h file, where I commented out the line:

  #if HOTENDS == 1
   // #define FAN1_PIN       MOSFET_D_PIN
  #else

Is this normal, or did I missed something else in my config, so that I have just 1 part fan and be able to use two extruders?!

InsanityAutomation, 2018-12-22:
Some of the newer code allows a single nozzle to store temps and fan speeds across tool changes for the opposing material, so identical gcode can be run here and on an idex or chimera type machine. My assumption is that with a second fan defined, its not using the storage code but trying to control the other fan. Ill look into it a bit and see if we need to add a define with single nozzle to always use fan 0.

lulu3d, 2018-12-22:
OK , Thanks for looking into it. ;-)

lulu3d, 2019-01-01:
OK, thanks for the fix It worked for me.

closed 2019-01-01 by lulu3d

thinkyhead, 2019-01-03:
> so identical gcode can be run here and on an idex or chimera type machine

G-code should always be tailored to the specific machine.

InsanityAutomation, 2019-01-03:
@thinkyhead in theory yes for best results, but in this case just mean that it functionally acts the same regardless of hardware and doesn't require crazy scripts or strange slicer hacks. It can doesn't mean it should :)