MarlinFirmware/Marlin #12693: Re-Arm Q's

boelle, 2018-12-20:
Hi all

I have come to the conclusion that re-arm is the best bet if i have a good setup allready

But a 2 Q's:

1) will it be able to run 5 steppers ( **TMC2208** ) and full graph display and still have 3 servo pin free?

     I use the servo pins to drive fans, one to cool of the bed after print so i dont have to wait. one to 
     drive the heatsink fan and one for electronics fan. so they do not have to be PWM but  i had a fan 
     extender that made it easy to connect

2) are there any sellers in europe? 
       i have asked panacut this but so far no answer. i could order direct but then again if they dont 
       answer the confidence level goes through the floor and if they ship it with USPS i'm sure it will be 
       delayed and when it gets here customs will take at least 2 weeks before they process it. UPS can get 
       it here in 2-3 days, granted it cost more but i dont care. i get what i pay for and i want fast shipping 
       and the customs stuff is taken care of while in transit

Bob-the-Kuhn, 2018-12-20:
Any pin on the LPC1768/69 boards can be a PWM.  If the desired pin isn't tied to the hardware PWM then a software PWM is used.

boelle, 2018-12-20:
so yes to all in 1? ie 5 TMC2208 and  full graph and 3 pwm (either hardware or soft)

then i still need to find a seller in europe

teemuatlut, 2018-12-20:
Theoretically TMC2208 should be supported on LPC platform but I don't think anyone has confirmed it yet.

Roxy-3D, 2018-12-20:
> so the q would be if there at least are enough free pins once a graph lcd is connected

There are extra pins for the ethernet and serial ports that can be used for other things.  (I had to do that to get the Max7219 debug matrix working on Re-Arm.)   Some of the pins on the RAMPS board do not have anything connected to them because the processor does not have an abundance of GPIO pins.

> i read somewhere a few hours ago that the re-arm uses the internal sd card to store what would have gone in the eeprom. but i will read more about it in a bit

Yes.   Most of the 32-bit processors do not have EEPROM capability built in.   And the Re-Arm board does not have any EEPROM.    But it does have a SD-Memory card slot, so we have been using that to provide the required storage of settings.    The SD-Memory card slot on the LCD Panel is used to hold the GCode if you wish to print from an SD-Memory card.

boelle, 2018-12-20:
oki, then i will only have to look up if there are 10 connected free pins

i might be a stretch but i would like to keep dual z drivers and endstops,

Roxy-3D, 2018-12-20:
[Re-ARM pinouts.pdf](https://github.com/MarlinFirmware/Marlin/files/2700453/Re-ARM.pinouts.pdf)

[Re_ARM_Schematic.pdf](https://github.com/MarlinFirmware/Marlin/files/2700454/Re_ARM_Schematic.pdf)

boelle, 2018-12-20:
i will,  i will use the sd card to print from as i have tested with printing from a pi many times and its just not as good. the pi stays but only using it for remote monitoring

now i just need to hunt down a seller in europe if possible, if not matterhackers is the best choice so far

closed 2018-12-20 by boelle

boelle, 2018-12-25:
so one re-arm on the way with dhl

with the mega board i used flash firmware via an octoprint plugin, but is it not something about that with the re-arm the internal sd-card shows up as a drive to a connected pc(and the pi) and you just place the bin/hex file on that drive and restart it?

ultimate my q is how do i update/flash firmware remotely via the pi/octoprint