MarlinFirmware/Marlin #12694: Error mode duplication positioning

xero83, 2018-12-20:
I have a big problem using my 3d machine with dual extruder.
When using the duplication mode, when doing the G28 does the first probe, but when doing the second one it does not go up and the extruder 2 tries to position itself colliding with the bed.
I use marlin 1.02, and the boot gcode is as follows:
M605 S2 X110
G21 ;metric values
G90 ; absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
M104 S {material_print_temperature} T0; sets extruders to print temperature
M109 S {material_print_temperature} T1; sets extruders to print temperature
M605 S2 X110
G28
G1 Z2 F1000
G1 X50 F2000
G1 Z-2 F1000
G92 E0 ;zero the extruded length

The idea is that, It is positioned once the G28 is done and produces the shock.
If I try to put the order M605 S2 X110 at the end, it does not do the duplication mode.

Any help?

Translated with www.DeepL.com/Translator

Roxy-3D, 2018-12-21:
Are you using a remote host like PronterFace or Repetier-Host?    Do you have extruder 1 active when you start your print?     I may have noticed an issue with this.      If this is the case...   Make extruder 0 active, and then start the print.   That may resolve the conflict well enough you can print while we look deeper into the root cause of the problem.

xero83, 2018-12-22:
the only way I've had to fix it is by doing a rehome, I send the code gcode:
M605 S2 X110 
G21; valores métricos 
G90; posicionamiento absoluto 
M82; coloque la extrusora en modo absoluto 
M107; comience con el ventilador apagado 
M104 S {material_print_temperature} T0; ajusta las extrusoras a la temperatura de impresión 
M109 S {material_print_temperature} T1; ajusta las extrusoras a la temperatura de impresión 
G28 
M605 S2 X110 
G28 X0
G1 Z5 F200
G1 X50 F2000 
G92 E0; cero la longitud extruida

and now if you do the function correctly.

InsanityAutomation, 2018-12-22:
At least you have a workaround now. No code is going to be changed on a version that old, And I dont believe the behavior is the same on the current branches. If you are not planning to validate a current 2.0 snapshot as well, you can go ahead and close this issue.

boelle, 2019-02-21:
@xero83 problem solved?

closed 2019-03-03 by thinkyhead