MarlinFirmware/Marlin #12696: filament sensor does not work marlin2.0 solution on pin 2 xmax

stefzouille, 2018-12-20:
hello,
 for those who do not manage to operate the filament sensor I put the sensor on XMAX pin and put the line: #define FIL_RUNOUT1_PIN 2 in the end 👍 
#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define NUM_RUNOUT_SENSORS   1     // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
  #define FIL_RUNOUT_INVERTING true // set to true to invert the logic of the sensor.
  #define FIL_RUNOUT_PULLUP          // Use internal pullup for filament runout pins.
  //#define FIL_RUNOUT_PULLDOWN      // Use internal pulldown for filament runout pins.
  #define FILAMENT_RUNOUT_SCRIPT "M600"
  #define FIL_RUNOUT1_PIN 2

then in Marlin-bugfix-2.0.x \ Marlin \ src \ pins edit the file pin_ramps.h:
 // define digital pin 4 for the filament runout sensor. Use the RAMPS 1.4 digital input 4 on the servos connector
#define FIL_RUNOUT_PIN 4
 and put 2 in place of 4 and it is good.

boelle, 2019-02-21:
@stefzouille problem fixed?

stefzouille, 2019-02-21:
It’s not a problem ,it’s a solution ,i give it..lol☺️

boelle, 2019-02-22:
submit a PR to bugfix 2.0 and click green close button below

closed 2019-02-22 by stefzouille