MarlinFirmware/Marlin #12699: Question STM32 INSTALL

istyszy, 2018-12-20:
Hello, 

I am new to stm32 controllers, and some step by step guideline would be useful.
I manage to flash the marlin v2 firmware to my DUE but i want to try with STM32 too.

As i found online that i have to use cubeMX to configure my stm32 and then somehow bring marlin and the config together... i did not understand.
Can somebody give me more info. 
My plan is to make an adapter board for the Core407V STM32F407VET6 and BluePill so it will fit on a RAMPS 4due.

Thanks!

AletheianAlex, 2018-12-23:
I had an idea to make a highly documented project for the STM32 HAL a while back since I have tinkered with those for a while, but I am waiting for some features to mature, and to see if the direction shifts to a unified HAL.  BUT if it helps to get started: generally for development, I have personally been using a USB to serial programmer or STlink, and the STM32duino project bootloader and tools to manually upload the firmware bin file that is compiled in the .pioenvs folder. I have been using the terminal under Linux and MacOS, but there are Windows tools as well. Check out the STM32duino project starter guide if you are completely new to STM32

boelle, 2019-02-21:
@istyszy problem solved?

closed 2019-03-03 by thinkyhead