MarlinFirmware/Marlin #12704: POWER_LOSS_RECOVERY Autolevel before resume print

TB1405, 2018-12-21:
Hi 
i use marlin bugfixes 1.1.8. 
I enabled the POWER_LOSS_RECOVERY feature but before resume print the printer will do also home Z and probe the 9 points from the. but on the is the print. 
Isn´t POWER_LOSS_RECOVERY usable with autolevel ?

boelle, 2019-02-21:
use bugfix 2.0

TB1405, 2019-02-27:
Same problem

boelle, 2019-02-27:
maybe @thinkyhead can tell what is going on if you attach your 2 configs as a zip file

TB1405, 2019-02-27:
i hope so 
[ConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/2910819/ConfigFiles.zip)

TB1405, 2019-02-27:
Maybe it isnt actived. But i had check it and it also doesnt work

thinkyhead, 2019-03-03:
It sounds like your G-code contains `G28` and `G29` and the power-loss resume is resuming the print from the beginning of the file instead of the last-recorded point. You should do a test while in dry-run mode and see if it restarts from the beginning or from some other point. You could use a simple gcode like this for testing:
```gcode
M300 S800 P500
M300 S600 P500
G28
G29
G1 Z5 F3000
G1 X80 Y80 F3600
G1 X120
G1 Y120
G1 X80
G1 Y80
G1 Z5.2
G1 X120
G1 Y120
G1 X80
G1 Y80
G1 Z5.3
G1 X120
G1 Y120
G1 X80
G1 Y80
G1 Z5.4
G1 X120
G1 Y120
G1 X80
G1 Y80
G1 Z5.5
G1 X120
G1 Y120
G1 X80
G1 Y80
G1 Z5.6
G1 X120
G1 Y120
G1 X80
G1 Y80
G1 Z5.7
; . . . and so on
```
If you hear the beeps and get the homing and leveling when you RESUME the print, it could mean a problem with the BIN file contents.

To debug further, add this line to your configuration and re-flash:
```cpp
#define DEBUG_POWER_LOSS_RECOVERY
```
If you are connected to a host or serial terminal while doing print, power-loss, and resume, you will get lots of extra output in the host console. It might point to the cause of the problem.

TB1405, 2019-03-03:
Yes you are right in my gcode is at the beginning g28 and g29 for autolevel function but why is this a Problem? 
I will chech this and Report later.

thinkyhead, 2019-03-03:
>  but why is this a Problem?

Indeed, that is the question.

TB1405, 2019-03-14:
today i checked it with the newest 2.0.x  
this is the ouput of serial monitor after powerloss 
`Connecting...
start
Printer is now online.
echo:Marlin bugfix-2.0.x
echo: Last Updated: 2018-01-20 | Author: (none, default config)
echo:Compiled: Mar 14 2019
echo: Free Memory: 2692  PlannerBufferBytes: 1376
Error:EEPROM datasize error.
[ERROR] Error:EEPROM datasize error.

echo:Index: 599 Size: 571
echo:Hardcoded Default Settings Loaded
echo:SD card ok
Load Job Recovery Info...
valid_head:1 valid_foot:1
current_position: 210.00,199.00,10.35,0.00
feedrate: 1500
target_temperature: 215
target_temperature_bed: 60
fan_speed: 0
leveling: 1
 fade: 0
cmd_queue_index_r: 1
commands_in_queue: 4
> G92 E0
> G0 X2 Y-4 Z0.2 F9000
> 92 E0
> G1 X80 E25 F750
sd_filename: /CFFFP_~2.GCO
sdpos: 462
print_job_elapsed: 139
---
echo:enqueueing "M1000 S"
Testing X connection... OK
Testing Y connection... OK
echo:SD card ok
echo:enqueueing "M1000"
echo:Bed Leveling OFF
echo:Fade Height OFF
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing`

there was no beep before or after power loss

5a2v0, 2019-05-16:
Hi all, same problem here.
Geeetech i3 Pro B (gt2560 A+) - Marlin 2.0 bugfix (downloaded on 2019/02/21).

After a power-loss recovery, the printer heats bed and extruder and starts with g29 operation with the object on the bed!
And if the printer goes off again, after the recovery it will not shows the resume menu.

Now i'm printing, but I downloaded today the bugfix fresh version so i will try soon

robbycandra, 2019-05-16:
@5a2v0 , try newest marlin2.0..  The problem have been fixed.

closed 2019-07-10 by shitcreek