MarlinFirmware/Marlin #12707: Marlin 2.0 vs reprapFirmware. Clash of the Titans

vladimirRUS007, 2018-12-21:
Marlin 2.0 and reprapFirmware.
Board used RURAMPS, Arduino DUE.
Is the firmware stable and productive for the Arduino DUE compared to the 32 bit reprapFirmware.
Interested in functionality and performance. And how will this affect print quality?

Roxy-3D, 2018-12-21:
There is no clash.    People are free to use (and contribute code) where ever they want.

Your title is designed to stir up hard feeling and polarize the community.  We don't need that here.

closed 2018-12-21 by Roxy-3D

vladimirRUS007, 2018-12-21:
Which one is better here's what I wanted to know

vladimirRUS007, 2018-12-21:
I'll go to the reprap firmware branch, I'll write the same. People should know what are the pros and cons of each of them. Am I right??? answer me please