MarlinFirmware/Marlin #12711: Prusa went crazy when using larger extruding speed than the extruder moving speed

thisishaoliang, 2018-12-22:
Due to our application, we have to use a larger extruding speed than the extruder moving speed on Prusa i3. Please check the printing process here [https://youtu.be/mt1rG_EnwkA](url)

I am using an extruder for 1.75 mm filament which contains a gear box (5.65:1 gear ratio), and a 1 mm nozzle. The marlin firmware is based on 1.0.2 version.

[image]

The printing path is shown in the picture above. During the whole printing process, i.e. the extruder moves from point A to point B, the nozzle height is set at 10 mm, the extruding speed (filament feeding speed) is kept at 300 mm/min in the gcode, the extruder moving speed is set at 262.5 mm/min.

I gave the description of the problem in the following texts, as well as in this video: [https://youtu.be/dekZ-kOLgCI](url)

Under these settings, the extruder always pauses at point C, while still keeping extruding. After a few seconds, the printer seems losing its mind, moving towards point D, then F, which is not asked by the gcode. The extruder couldn't even stop when it hits the x axis endstop sensor. I have to turn off the printer to avoid further damage.

However, after changing the extruding speed to 800 mm/min, and the extruder moving speed to 700 mm/min. (their ratio is the same, 300 / 262.5 = 800 / 700) The extruder is able to finish the whole process requested by the gcode, i.e. moving from point A to point B without any pauses or mistakes.

Comparing the two Gcode files in the attachments, the amount of material requested to extrude during the whole printing process is the same in both files. The difference is just the extruding speed and moving speed, which are both scaled up. (The extruding speed in the Extruding_speed_800_mm_min.gcode actually is too high, making it impossible to extrude any material smoothly)

Then I'm guessing the issue is due to some kind of limitation in the firmware. I attached the two gcode files, my configuration.h and configuration_advance.h files.

Could anyone give me some hint about the reason for the printing mistakes with the Extruding_speed_300_mm_min.gcode file? Thank you very much.

[github.zip](https://github.com/MarlinFirmware/Marlin/files/2704637/github.zip)

boelle, 2019-02-21:
is this a marlin issue?

thisishaoliang, 2019-02-21:
> is this a marlin issue?

Hi Boelle,
I still have no idea now. It could also be a hardware issue.
At least there is no problem with the g-code, cause I tried on Raise3D, it worked perfectly.

boelle, 2019-02-21:
so is this a problem on a prusa printer?

thisishaoliang, 2019-02-21:
> so is this a problem on a prusa printer?

Yes, there must be a problem on this prusa in our lab.
This happens a lot, till some point during the printing, the extruder moves differently from what the g-code asks.

thisishaoliang, 2019-02-21:
> so is this a problem on a prusa printer?

The board is Rumba.
I was wondering if the problem is from the board. I am not sure if I need to change a new one to see.

boelle, 2019-02-21:
contact prusa customer support

thisishaoliang, 2019-02-21:
> contact prusa customer support

Surprising that I never contact them... I'll try, thanks

boelle, 2019-02-21:
please click green close button also

closed 2019-02-21 by thisishaoliang