MarlinFirmware/Marlin #12712: [1.1.9] LCD cycles in and out of different menus, when heat bed is turned on using LCD/ pronterface.

mhsuhail, 2018-12-22:
Hi,
I recently installed Marlin 1.1.9 on a Prusa i3 clone, and I have been having some issues using the LCD panel after turning heat bed on while a print is in progress. 
If I try to turn the heat bed on after a print has started, the LCD panel cycles between different menus rendering it useless/difficult to use. I am using RAMPS 1.4 setup and have used several different LCD (2004 as well as 12864) panels without getting rid of the problem. The same thing happens if I try to start a heat bed using pronterface amidst a print. 
Below is a link to the video highlighting the issue, I am also attaching my configuration files.
https://www.youtube.com/watch?v=FA0ixLi4_p4.
Thanks 
Regards
Misba
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2704688/Configuration.zip)

boelle, 2019-02-21:
problem solved?

mhsuhail, 2019-02-21:
No, still there, just not using the heated bed for the time being. You facing similar issue??

boelle, 2019-02-21:
are you using bugfix 2.0?

mhsuhail, 2019-02-21:
I am running Marlin 1.1.8 and not 2.0.

boelle, 2019-02-21:
use 2.0

mhsuhail, 2019-02-21:
I tried to use it in December but had issues with auto bed levelling, will give it a go again. On the other hand I did not come across this issue in any of the forums or Github ISSUES page, so I am guessing that if it is a bug it might not have been resolved in later versions of Marlin. I might be wrong. Anyways thanks for taking time out for replying to my query.

boelle, 2019-02-21:
please click green close button below, 

you can open new issue if problem is still there

closed 2019-02-21 by mhsuhail

Fr33man85, 2019-05-07:
i have excatly the same bug. 

I own an Creality Ender 5 Printer and tryed some different Firmwares like TH3D, Stock Creality, Stock Marlin 1.1.9 and all ended up with the same Bug:
When Bed-Heating is on, the LCD jump´s into the Menu, and back to Info, back in Menu, back to Info,.... More or less fast, depending on - i think - the PID-Value calculated for the Cycle-time of the heating-element.

Only the Firmware, what was delivered, dont show this behavior. But this firmware is gone after the Update of the offered Stock Creality FW

Any ideas, why this happend?

greetings from Germany

Manuel

mhsuhail, 2019-05-08:
Hi Manuel, 
Sorry to hear about this. I was not able to solve it, nor was I able to find any documented case. Am not tinkering with my printer for the time being because of being busy with something else. Please let me know as well if you find a solution. Sorry again couldn't provide any help.