MarlinFirmware/Marlin #12718: Probe accuracy issue

Lyr3x, 2018-12-23:
I have a huge problem with my new inductive probe on my Hevo. I cant level the bed accurate because there is a huge difference after each probe:

M48 Z-Probe Repeatability Test
1 of 10: z: 0.042
2 of 10: z: 0.089
3 of 10: z: 0.119
4 of 10: z: 0.085
5 of 10: z: 0.142
6 of 10: z: 0.169
7 of 10: z: 0.233
8 of 10: z: 0.259
9 of 10: z: 0.283
10 of 10: z: 0.311
Finished!
Mean: 0.173125 Min: 0.042 Max: 0.311 Range: 0.269
Standard Deviation: 0.088278

Any ideas whats the problem here? Using the latest Marlin-2.0-bugfix on a Einsy board with an inductive 8mm probe.

Roxy-3D, 2018-12-23:
M48 probes the same spot again and again.  It measures the standard deviation of the value returned by the probe.    It is pretty clear, your probe is not measuring reliably.

Some people get inductive probes to work well.   But most people have trouble.    You might want to switch to a different probe.    Perhaps a higher quality inductive probe.   Or maybe a micro-switch based probe.

epatel, 2018-12-23:
Thomas made a good vid.
https://youtu.be/il9bNWn66BY

Lyr3x, 2018-12-23:
@Roxy-3D yep thank you. I thought maybe there is another point of failure i am not afraid of. I think i am switching the inductive probe to a genuine bltouch next year.

closed 2018-12-23 by Lyr3x

Bob-the-Kuhn, 2018-12-23:
Almost all your readings are increasing.  I've seen that when I was loosing steps.

Lyr3x, 2018-12-23:
Already checked that as I thought the same way first. The z a is is very accurate.