MarlinFirmware/Marlin #12723: Small, unexpected (but predictable) pauses when [un]retracting

VanessaE, 2018-12-24:
This is bugfix-2.0.x, commit 7d15854514f8e3818722eaf87dc96e098895ed79

### Description

When retracting via G10, a very brief pause (a few tenths of a second) happens with each extruder movement. That is, this pattern of movements happens:

draw a print line and stop -> retract -> brief pause -> travel and stop -> unretract -> start printing the next line

It's hard to tell, but I think I'm seeing pauses with G11/unretract as well, but they're much shorter.

It's considerably worse in sections of the print with lots of [un]retracts around short travels and short extrusions.

I was at 902167fa78a4893560d82dab8905c94ed265ed2c before, and this was not happening then.  The only significant configuration change I've made since then was to enable backlash compensation, but these little pauses seem to be present with or without that feature enabled (by which I mean compensation amounts set to some values other than 0).

### Steps to Reproduce

Print something with a lot of retracts :wink: 

**Expected behavior:** [What you expect to happen]

end drawing a print line and stop -> retract -> immediately travel and stop -> unretract -> start printing the next line

#### Additional Information

configs:  [Marlin-configs-20181224.zip](https://github.com/MarlinFirmware/Marlin/files/2707081/Marlin-configs-20181224.zip)

VanessaE, 2019-01-08:
Still present at commit da42f27c9e6ffe5bf2501c78ad23edf7a82e9f35...  Just to make sure I'm not seeing things, I told Slic3r to not to use FW retract (thus normal G1 Exxx moves to perform a retract), and that way works exactly as it should.

VanessaE, 2019-02-11:
No comment on this?  FW retraction is all but unusable due to this glitch?

boelle, 2019-02-21:
also a problem if you download todays copy of bugfix 2.0?

VanessaE, 2019-02-22:
I've just updated to commit f6ab62bc (HEAD as of this moment).

This issue still exists.

mrvanes, 2019-06-21:
I can confirm I see the same (un)retract pauses coming from 1.1.5.70 to 2.0.0.135 (Lulzbot fork of bugfix-2.0.x). The latest merged commit is 0f04ba from InsanityAutomation on may 28th.

boelle, 2019-06-22:
we need a label on this one

mrvanes, 2019-06-26:
It seems the planner.synchronize() are quite expensive in FWRetract::retract. Without them, I see the expected behaviour. I made a small video to show the effect and difference without planner.synchronize()
https://youtu.be/YVR90nZpxIY

VanessaE, 2019-06-26:
Thank you for posting that video, @mrvanes.  While I can't speak to the code change you mentioned, the video shows the issue perfectly.  Why didn't I think of that? :stuck_out_tongue:

shitcreek, 2019-07-10:
Should we consider this issue resolved? Bug fixed?

closed 2019-07-10 by VanessaE

thinkyhead, 2019-07-11:
> It seems the planner.synchronize() are quite expensive in FWRetract::retract.

Indeed. This is why I added so-called "sync blocks" to the planner. So we can modify the current XYZE position without needing to pause the planner.