MarlinFirmware/Marlin #12724: Enabling Z-lift/hop adds extruder noise (affects its acceleration?)

VanessaE, 2018-12-24:
### Description

This is bugfix-2.0.x, commit 7d15854514f8e3818722eaf87dc96e098895ed79

When using Z-hop/lift as part of firmware retract, retractions have occasional "clash" sounds, as if the extruder is jumping to full speed without acceleration (the default of which is 2000 mm/s², max is 10000 mm/s²).  I don't hear it all the time, and it's probably only noticeable on geared extruders (i.e. the sudden motion amplifies the normal noise from gear backlash).

If Z-hop is not enabled, then retractions behave normally.

This didn't occur with commit 902167f .

### Steps to Reproduce

Print something with lots of retractions.  Issue let's say, `M207 Z0.6`, and listen for the difference in extruder behavior.

**Expected behavior:**

No excess extruder noise just because I turn on Z-lift :smiley:

Configs: [Marlin-configs-20181224.zip](https://github.com/MarlinFirmware/Marlin/files/2707326/Marlin-configs-20181224.zip)

boelle, 2019-02-21:
also a problem with todays copy of bugfix 2.0?

VanessaE, 2019-02-22:
This doesn't seem to be a problem anymore.

closed 2019-02-22 by VanessaE