MarlinFirmware/Marlin #12725: [bugfix 1.1x] Mini RAMBo, UBL fails to probe mesh with garbage queue commands

Dolnor, 2018-12-25:
### Description

Running bugfix 1.1.x on a Mini RAMBo board. Am trying to switch from Bilinear leveling with a BLTouch to UBL and failing at that. Not sure if it's a board problem or a config problem, but thus far it's looking like the firmware is not playing along.  

### Steps to Reproduce

1. Enable AUTO_BED_LEVELING_UBL with 4 grid mesh points (ideally, I'd want it as 5x4, but start small) 
2. Through UBL menu in Prepare (or host), home with G28 and take cold mesh with G29 P1
3. Printer resets itself either after probing one point or right away after homing completes and is confirmed by the buzzer chirping

**Expected behaviour:**

The mesh levelling to carry on through the grid configured and record the mesh map in selected slot.

**Actual behaviour:** 

In most case either the motors will start to hum and do nothing; it will probe one mesh point and reset randomly without any console output or the printer will reset right away. 

Enabling debugging for G28 and G29, I can see the homing sequence completes successfully but the moment G29 P1 is executed, putting 'Doing G29' on LCD, in 95% cases serial console will come back with something random along the lines of:

> SENDING:G29 P1
> Default storage slot 0 selected.
> Mesh invalidated. Probing mesh.
> echo:busy: processing
> echo:enqueueing "faQCK]ÁE§A	=˜q91@6Û"

> SENDING:G29 P1
> Default storage slot 0 selected.
> Mesh invalidated. Probing mesh.
> echo:busy: processing
> echo:enqueueing "”;2
> ”Úƒ”;2”;2”;2”;2”;2”"
> Attempted to write invalid text to console, which could be due to an invalid baudrate

> SENDING:G29 P1
> Mesh invalidated. Probing mesh.
> echo:busy: processing
> echo:enqueueing "4"
> Bed Topography Report for LCD:
> NANNANNANNAN
> NANNANNANNAN
> NANNANNANNAN
> NANNANNANNAN
> echo:Unknown command: "G29 P1"

> SENDING:G29 P1
> Mesh invalidated. Probing mesh.
>   current_position=(123.00, 77.00, 5.00) : set_probe_deployed
> deploy: 1
> echo:busy: processing
> echo:enqueueing "”"
> echo:enqueueing "2”;2”;2”;2”;2”;2”;2”;2"
> echo:busy: processing
>       probe_pt(118.00, 100.67, raise, 0, probe_relative)
>   current_position=(123.00, 77.00, 11.45) :

This will cause and immediate printer reset. I've tried changing the baudrate from 115200 to 250000 as indicated by one of the bogus serial data command verbose messages generated by Pronterface, to no avail. 

#### Additional Information

Configuration files and board pins config attached:
[mini_rambo_config.zip](https://github.com/MarlinFirmware/Marlin/files/2708603/mini_rambo_config.zip)

Dolnor, 2018-12-25:
With all the debugging bells and whistles enabled, it comes back with 

> Sketch uses 232462 bytes (90%) of program storage space. Maximum is 258048 bytes.
> Global variables use 5069 bytes of dynamic memory.

I'm not sure how much SRAM the board has to be honest. Without DEBUG_LEVELING_FEATURE, it's just 60% of ROM being taken up by Marlin. Says I've enough memory to store 41 meshes. 

I've spotted my maximum feedrates were through the roof, so I've groomed them down to 100 on XY, but the problem is still there. The only way to use UBL for now is to manually define a mesh and fill it in with 0.5s, then go and drag a paper under every mesh point and save it so.

Roxy-3D, 2018-12-25:
There was a recent problem with UBL's G29 P1 not working and it turned out to be memory problem.

Can you try turning of SD-Memory card support?   //#define SDSUPPORT

That will free up a lot of RAM and Program memory.    And then see if you can do a G29 P1 and have it complete.   If that does complete, you can probably save the mesh, and turn SDSUPPORT back on (because you shouldn't need to re-probe the bed).

Dolnor, 2018-12-25:
Yup, disabling SDSUPPORT allows the probe to function properly with G29 P1. Cheers

I then hit a problem when multi-probing a 5x4 mesh having BLTouch not responding and going up-up-up the Z axis once probing a point fails (for some reason towards the left of the bed), which I've not had with Bilinear levelling. Reducing the mesh to 4x4 seems to probe fine, but it's pretty sparse like that.

It's not a genuine probe (I have one coming from Filafarm) but a TL-Touch. If you've got any pointers for that, I'd appreciate it. If we are venturing out of topic for this issue report, I'm hopefully going to figure that out or leave alone until a genuine probe comes.

Happy Christmas!

Roxy-3D, 2018-12-25:
The up, up, up problem is usually associated with going off the mesh.   But that shouldn't be happening while building the mesh.     How much memory (Both RAM and Program space) do you have when you build the firmware?

Dolnor, 2019-01-01:
Sorry, completely forgot about the printer during the holiday season.

I'd posted a similar line above on prior request, but I've changed a few things since that message, so we are not at these figures.
> Sketch uses 154860 bytes (60%) of program storage space. Maximum is 258048 bytes.
> Global variables use 7534 bytes of dynamic memory.

UBL kept being finicky to set up, the probe would either drift along a probing point without even attempting to measure, would actually measure or would carry on going up-up-up indefinitely. So building a mesh was a royal PITA. Tried changing the insets, tried modifying the mesh bounds all to no avail.

Ended up switching to regular 5x5 bilinear method for the time being. Seems to work alright, however it also appears to be having clashes (more annoying that UBL) with SDSUPPORT when activated with M420 S1, which UBL didn't seem to have. Clashes being the LCD (discount reprap controller) freezing completely at random points during SD print if bed levelling was enabled. The encoder will either keep providing the feedback sound or not, but LCD stops responding completely while the print is going as if nothing happened. Now, I should probably also mention that I can't get SD to initialize properly with alphabetical sort order enabled any other way but by using SDSORT_USES_RAM. If I don't have it configured like so, no matter what SD card I try to use, the printer will keep resetting with SD init fail. 

I should probably create another issue for that one, you think? And we can close this one about UBL as there's a known workaround to build the mesh at least?

Cheers and happy new years!

boelle, 2019-02-21:
@Dolnor problem solved?

closed 2019-03-03 by thinkyhead