MarlinFirmware/Marlin #12726: Z homing with endstop and BLTouch

DerDominik, 2018-12-25:
Hi there,

I'm using an Anycubic i3 Mega, which has 2 Z-Endstops for adjusting x-axis and I've also built a BLTouch sensor, which actual only works for mesh leveling (which works).

In my idea I want to to a  dual Z homing:
First it should home X/Y and then Z to the endstop (for x-axis adjusting) (works actual with G28).
Then it should go to X110 and Y110, and do a Z homing with BLTouch.
It is possible to switch the z homing method from endstop to BLTouch?
Both signals (endstop and BLTouch) goes to the same pin.

Thanks for your help :-)

TAz00, 2018-12-25:
You'd have to use seperate pins for bltouch and endstop. 
I dont use a YMax endstop, so I used that pin# for probe pin.  (looked in at board pins to find YMax pin #)

DerDominik, 2018-12-26:
It must be completely different pins?
I checked the wiring, and BLTouch signal is going to X+ Endstop.

In my mind it could be realized like this:
If X and Y is not homed or X=0 and Y=0 -> Classic Z-Homing which reacts only on Z-Endstop
If X and Y > defined value (let's say middle of the bed) -> Z-Homing with BLtouch

boelle, 2019-02-21:
@DerDominik problem solved?

closed 2019-03-03 by thinkyhead