MarlinFirmware/Marlin #12727: bugfix-2.0.x dual_Z endstop and probing

TAz00, 2018-12-25:
### Description

I was having the same problem as this guy : https://github.com/MarlinFirmware/Marlin/issues/11134
But I have dual_z endstops and use YMax pin for probe. 

I'm using 1.1.9, but I've checked the code for bugfix 2.0.x and the "bug" is there aswell.

### Steps to Reproduce

https://github.com/MarlinFirmware/Marlin/issues/11134

### Suggested Fix

Expand the previous commit https://github.com/MarlinFirmware/Marlin/commit/485d50ecb08b0856863a2dfe558dee284463a23f to also cover DUAL_Z_ENDSTOPS

        #if ENABLED(Z_DUAL_ENDSTOPS)
          #if ENABLED(Z_MIN_PROBE_ENDSTOP)  
            if (!z_probe_enabled) PROCESS_DUAL_ENDSTOP(Z, Z2, MIN);;
          #else
            PROCESS_DUAL_ENDSTOP(Z, Z2, MIN);
          #endif
        #else
          #if ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
            if (z_probe_enabled) PROCESS_ENDSTOP(Z, MIN);
          #elif ENABLED(Z_MIN_PROBE_ENDSTOP)
            if (!z_probe_enabled) PROCESS_ENDSTOP(Z, MIN);
          #else
            PROCESS_ENDSTOP(Z, MIN);
          #3766 
        #endif

boelle, 2019-02-21:
@TAz00 submit PR to bugfix 2.0

closed 2019-03-03 by thinkyhead

kulfuerst, 2019-05-23:
The code, as it is standing here, only works in Marlin 1.1.9. To make it work in Marlin2 slight changes have to be made.

        #elif ENABLED(Z_DUAL_ENDSTOPS)
          #if USES_Z_MIN_PROBE_ENDSTOP  
            if (!z_probe_enabled) PROCESS_DUAL_ENDSTOP(Z, Z2, MIN);;
          #else
            PROCESS_DUAL_ENDSTOP(Z, Z2, MIN);
          #endif
        #else
          #if ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
            if (z_probe_enabled) PROCESS_ENDSTOP(Z, MIN);
          #elif USES_Z_MIN_PROBE_ENDSTOP)
            if (!z_probe_enabled) PROCESS_ENDSTOP(Z, MIN);
          #else
            PROCESS_ENDSTOP(Z, MIN);
          #endif
        #endif