MarlinFirmware/Marlin #12728: [2.0.x] G28 command is broken, doesn´t take Y in combination with X or Z

Itox001, 2018-12-25:
Today I had a very large and time critical 40 hour print job fail 20 hours in because of a bad bowden coupler at 4 AM. So I pause the printer (advanced pause yay) and proceed to fix the extruder. Then I realize that since a minute passed before I spotted it, half a layer is missing, so I can't just resume the print...

Ok, no problem, I'll see where it stopped extruding and start the Gcode from there, I've done it several times. For this I will connect the computer to the printer which will cause a reboot and position will be lost. This is ok because as long as I don't touch anything, the printer shouldn't move more than 8 microsteps when the steppers power up again, specially the Z since that can't be homed with a full print bed.

I then cut the gcode to where it left and make the neccesary modifications for it to work standalone, such as G92 to actual Z position, correct feedrate for the move, etc. One of this modifications is to home X and Y (in hindsight, probably should have left it alone...) but not Z, since it would crash with the print. So I try to do a G28 X Y to home just the X and Y axis and this is where the bug showed:

**G28 X Y only homed the X axis. It completely ignored the Y.**

So this messed with my positioning, which led to a very frustrating time getting everything ready again and trying to resume the print. G28 Y works fine, but in combination with any other axis, the Y is ignored. G28 X Y Z will only home X and Z and G28 X Y will only home X. Ended up using G28 X followed by G28 Y. 

On another note, isn't there any way to avoid the printer from raising Z when homing? This also caused me some grief and I could't find any way to disable it, the only relevant line is //#define Z_HOMING_HEIGHT 4, but thats commented out and is 4, when in reality the Z rises 5 mm. Googling I found that this setting was exposed before with Z_RAISE_WHEN HOMING or something like that, but it's no longer present.

Sorry for the long rant, I only slept 1 hour today becuase of this :) Merry christmas.

That printer is using Marlin 2.0.x from September, but issue is also present on my other printer with 2.0.x from December 8th, so I would expect it to be also present on the latest one, since I found no related commits since then. Sorry for not testing it on the latest, but gotta sleep some...

Attached are the config files.
[G28 X Y bug.zip](https://github.com/MarlinFirmware/Marlin/files/2708904/G28.X.Y.bug.zip)

FanDjango, 2018-12-25:
I just tried a "G28 X Y" and can see both being homed simultaneously. I am on the current 2.0.x. Are you sure no program is messing with your GCODE before it reaches the printer? With what are you entering the G28 X Y, have you got a debug log (like serial.log in OctoPrint?). Have you tried G28 XY? I looked at your config and can't see anything bad.

Itox001, 2018-12-25:
I´m running the printer from USB with Repetier Host, sending the command with the command box:
[image]
This is on the 2.0.x Dec 8th build, the other machine (Sept build) that presented the same issue was also via Repetier Host but with a Gcode file.

Do you think this could be a Repetier Host issue? I could try from SD, when the 20 hour print finishes and frees up my SD.

FanDjango, 2018-12-25:
Far below is my "off the cuff response" to your post.

But first, after some more thinking I would like to interject a question:

***That printer is using Marlin 2.0.x from September***
Unless you have not printed anything from September to last night, you should have noticed G28 being broken or do you not have a G28 with X and Y both in you start GCODE when slicing?

My start code is:
(And as you can see, there is a G28 X Y in there, which I can observe while it is working correctly)
```
; Printer Begin GCODE

M502 ; Set to Marlin defaults
M500 ; Store in EEPROM

M107 ; Fan off

G91   ; Relative

M140 S{material_bed_temperature_layer_0} ; nowait

M104 S150 ; nowait

G1 Z+25 F480   ; up a little for some safety before home XY
G28 X Y              ; Home XY at 50mm + unknown
G28 Z                 ; Home Z no. 1 and then...

G90 ; Absolute

G1 Z175 F720    ; go up quickly nearly all the way
M211 S0            ; turn off software endstops
G1 Z185 F120    ; and up slowly into physical limiter because of ANET A8 dual Z unsync danger
M211 S1            ; turn on software endstops
G1 Z15 F720      ; and down again quickly nearly all the way and then
G28 Z                 ; home Z again, now both Z motors are in sync

;G0 Z180 F720     ; raise nozzle and ...
;G0 X0  Y{machine_depth} F9600  ; present bed for cleaning
;here, insert pause command

G1 Z10 F720      ; Down to 1cm
G1 X3 Y3 F9600 ; {speed_travel} is mm/s, but need mm/min

M190 S{material_bed_temperature_layer_0} ;wait
G29
M420 S1 Z5    ; set Leveling fade height to 5 mm

M104 S{material_print_temperature_layer_0, 0} ;nowait

G1 X3 Y3 F9600                ; X=3 Y=3 inner corner
G1 Z0.10 F720                 ; Drop to bed while heating

M109 S{material_print_temperature_layer_0, 0} ;wait
M190 S{material_bed_temperature_layer_0} ;wait

M83 ; E relative
G92 ﻿E0.0                               ; reset extrusion distance
G1 X30      E+6.0 F1000.0    ; 20mm intro line @ 0.30
G1 X70      E+1.5 F1000.0    ; +40mm intro line @ 0.08
G1 X110    E+3.0 F1000.0    ; +40mm intro line ﻿@ 0.15
G1             E-1.0  F900          ; retract to avoid stringing
G1 X116    E0      F1000.0     ; +6mm intro line @ 0.00
G1 X111    E0      F1000.0     ; -5mm intro line @ 0.00﻿
G1 Z10                F480          ; Up to 1cm
G92 E0.0                               ; reset extrusion distance

M84 S600     ; steppers off if inactive longer than 10 minutes (safety)

M900 K0.8    ; Linear Advance Coefficient (experimental)
; turn off coasting in Cura !!!
; or set this M900 value to 0.0

M82 ; E absolute (currently, this is absolutely expected by CURA
```

Ok, here is the original answer I was writing:

I understand. Well, I do not think it is a Repetier Host issue. But whenever I am totally unsure of where the error may lie (if that were the case), I would connect the printer via USB to my PC and open up a serial terminal and type the commands in there, and see the response natively. Once having done that, one can begin to suspect the printer firmware.
Of course, you can go the other way, and this might be easier right now: Assuming that Repetier is not fooling around with G28, do the loop: try to position the nozzle (after successful homing) to X = 100, Y = 100 and Z = 100. Confirm visually. Use G1 commands to do that.  And then test the G28 variations systematically (also without blanks between XYZ), to see which work and which not. Make a table depicting all combinations. 
Then report this table  
It works fine on my config, so something is strange.

swilkens, 2018-12-26:
Make a simple GCODE file that only has something like:

> G28
> 
> G1 X 40 Y 40 Z 40 
> 
> G28 Y
> 
> G1 X 40 Y 40 Z 40
> 
> G28 X Y

And run if from the SD card. If that works properly you have isolated the issue. If that also fails we'll have to dig deeper.

Itox001, 2019-01-16:
Hi, @FanDjango and @swilkens, sorry for the delay.

I have made some tests and SD printing works fine with G28 XY or G28 X Y. Then, I made tests with a simple serial connection with Putty and that works fine too. So it is a Repetier Host problem apparently.

Damned Repetier. Is there any way to see what the controller is actually receiving from repetier so I can open a bug report there?

FanDjango, 2019-01-16:
Oof.

> I made tests with a simple serial connection with Putty

That was the best way.

Doesn't Repetier Host have somekind of serial logging capability? I dunno, I don't use it.

Itox001, 2019-01-16:
I also not know. But since this is not a Marlin issue, I´m closing it. 

Thanks for your assistance!

closed 2019-01-16 by Itox001