MarlinFirmware/Marlin #12730: Installed TMC2130 on a delta and now mechanical endstops are ignored when doing homing

Anycubic, 2018-12-26:
### Description

Installed TMC2130 on a delta and now mechanical endstops are ignored when doing homing

### Steps to Reproduce

1. Install TMC2130 on a delta printer
2. Perform G28 and the axis will crash against endstops ignoring them

**Expected behavior:** axis should stop and the display should show "0" "0" "height" as "x", "Y" and "Z"

**Actual behavior:** "?" is shown on "X", "Y" and "Z"

#### Additional Information

Tested endstops with M119 and they seem working right, I can see "open" and "triggered" when engaged. BTW I was coming from A4988 where was having no issues at all. I just changed the TMC part and X_CS, Y_CS and Z_CS stuff

[pack.zip](https://github.com/MarlinFirmware/Marlin/files/2710010/pack.zip)

Anycubic, 2018-12-26:
Ok, I spent another couple of hours trying to understand what's going on. I observed the following:

1) after issuing G28 command, every axle which hits his endstop first will stop for a while and then keeps on moving up (?). Of course it will keep on crashing on the endstop
2) if a second axle (any) will hit his endstop AFTER the first one, the first axle will stop trying to go up (like if recognised at last it actually hits his endstop), the second one will stop, also the third axle then will stop as it should after hitting his endstop.

I hope I made myself clear, I can't really understand what is going on and how using TMC2130 could have started this strange behaviour

Thanks

Glod76, 2018-12-27:
try changing these seems you need them at 0 for tmc2130
#define X_HOME_BUMP_MM 5
#define Y_HOME_BUMP_MM 5
#define Z_HOME_BUMP_MM 5

Glod76, 2018-12-27:
are you using the latest marlin build? try  the newest build and see if it's doing the same

Anycubic, 2018-12-27:
> try changing these seems you need them at 0 for tmc2130
> #define X_HOME_BUMP_MM 5
> #define Y_HOME_BUMP_MM 5
> #define Z_HOME_BUMP_MM 5

Already tried that, didn't work. Btw also bump is doing crazy thing, instead of bumping axis just go back. For sake of clarity, I'm not talking about sensorless homing

Anycubic, 2018-12-27:
> are you using the latest marlin build? try the newest build and see if it's doing the same

I'm using 5 November build, downloaded 20 December build and going to test with it

mcmuff91, 2019-02-07:
Hello Anycubic,

could you upload your config ? Having the same problem atm. started from the beginning a few time but couldnt find a solution.