MarlinFirmware/Marlin #12731: Creality Ender-4 bugfix-2.0.x Fan Speed M106 does not work

ShaggyDog18, 2018-12-26:
### Description
M106 does not work for Ender-4. Fan is not controllable.

### Steps to Reproduce

You added two new lines to the   pins_RAMPS_ENDER_4.h   file:
#define ENDER4_FAN_PIN RAMPS_D9_PIN
#undef FAN_PIN	
These lines were not present in the earlier versions
** Bug Fix
just comment new lines in the  pins_RAMPS_ENDER_4.h  file.

Ludy87, 2018-12-26:
If 'CASE_LIGHT_PIN ENDER4_FAN_PIN' is activated, the PINs conflict.

Both used RAMPS_D9_PIN

dsbragg1979, 2018-12-27:
If M106 is enabled as a parts fan a case light can still be used and controlled from G code.
If m106 is disabled in favor of a dedicated case light (which would be better served running from LSU on it's own switch) suddenly none of our current g-code works for the parts fan.
The ender4 comes stock with m106 used to enable case light/pwm fan and this is what its users are familiar with and setup for. Changing it is not only not stock bit screws up our g code whether it is used to turn the case light on while printing and off when done or , to control an add on parts fan(standard creality parts fan works)

I am a member of same group and this ,caused quite an upset, keeping many E4 users away from upgrading Marlin.  Thankfully Serigy hear our cries and fixed it  for us.
 TL/DR if it ain't broke font fix it M106 works great for case light or parts fan, changing it breaks our g-code and requires seperate files to share between printers.

boelle, 2019-02-21:
@ShaggyDog18 problem solved?

ShaggyDog18, 2019-02-23:
nope, it was not. 
I am still commenting new lines (mentioned above) in the RAMPS_ENDER_4.h file to make it work...

thinkyhead, 2019-03-03:
`M355` is the command to enable or disable the case light. Is that what Ender 4 users are also using? The case light may be connected to any free 12V fan or heater output, or use a 5V pin to a relay or MOSFET that passes 12V to the case light.

`M106` is the command to set the print cooling fan speed. If a light and fan happen to be on the same circuit, the light will come on with the fan.

We do check to make sure the case light (if enabled) doesn't share the `FAN_PIN`:
```cpp
/**
 * Test case light not using the same pin as the fan
 */
#if ENABLED(CASE_LIGHT_ENABLE) && CASE_LIGHT_PIN == FAN_PIN
  #error "You cannot set CASE_LIGHT_PIN equal to FAN_PIN."
#endif
```

This is what I would propose in `pins_ENDER_4.h`…
```cpp
. . .
#include "pins_RAMPS.h"

// The board only has one PWM fan connector. The others are 12V always-on.
// The default config uses this pin to control the brightness of the LED
// band (case light). Thus the hotend and controller fans are always-on.

#if ENABLED(CASE_LIGHT_ENABLE)
  #undef FAN_PIN
  #ifndef CASE_LIGHT_PIN
    #define CASE_LIGHT_PIN RAMPS_D9_PIN // Override the default pin if needed
  #endif
#endif
```

ShaggyDog18, 2019-03-03:
Good solution! Thank you!

boelle, 2019-03-03:
@ShaggyDog18 so problem fixed? or does it still need some work?

ShaggyDog18, 2019-03-03:
I guess you have to remove line 301 in config_adv.h - it is not necessary now..
 301:   //#define CASE_LIGHT_PIN ENDER4_FAN_PIN       // Override the default pin if needed

ShaggyDog18, 2019-03-03:
Though I did not test it yet, theoretically, should work fine. Yes, you may close the case. Thank you.

boelle, 2019-03-03:
> Override the default pin if needed

so if it works without removing // then everything should be ok

only you can close by clicking close button below

closed 2019-03-03 by thinkyhead