MarlinFirmware/Marlin #12732: [2.0.x] Compile errors on delta with TMC2130 and sensorless homing

vilsed, 2018-12-26:
Hello y'all and happy holidays! Holiday season is more or less the only free time I get to work on my delta machine, and unfortunately, I'm getting issues with the latest 2.0.x...

For a long time I have TMC2130 drivers installed and working, with the latest chopper settings update, I'd say, flawlessly. Although I was always hesitant of using the sensorless homing feature, as I thought that mechanical endstops are more precise (which is very important for a delta machine). Unfortunately, for the last five months or so (after updating to the latest version, five months ago), my endstops weren't working correctly. Sometimes the machine would fail homing before the print, sometimes after. And the failure mode was really peculiar too - all the carriages would start moving together, as they should, then the first to hit the endstop would, well, hit the endstop and stop, then the other two would bump and rebump their endstops as they should, but the first one would never rebump. It would just stay there, pressing the microswitch down. And VALIDATE_HOMING_ENDSTOPS would spit out a failure. I tried enabling the SW noise filtering, adding capacitors, nothing really helped. But all this information is besides the point, just a curiosity. I gave up, decided to try sensorless homing. And there the current issue lies.

### Description

On a delta kinematics machine, with Re-ARM board, TMC2130 drivers and SW_SPI enabled, enabling sensorless homing gives the following compilation errors:

`Marlin\src\module\delta.cpp:225:46: error: 'stepperX' was not declared in this scope`
`Marlin\src\module\delta.cpp:226:46: error: 'stepperY' was not declared in this scope`
`Marlin\src\module\delta.cpp:227:46: error: 'stepperZ' was not declared in this scope`

### Steps to Reproduce

1. Configure latest 2.0.x version for delta kinematics to use TMC2130 on SW SPI with sensorless homing;
2. Compile;
3. Get errors, cry.

**Expected behavior:** Marlin compiles and life is beautiful.

**Actual behavior:** Marlin does not compile.

#### Additional Information

I'm attaching my Conf.h and Conf_adv.h here, aswell as full compiler output.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2710260/Marlin.zip)

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip; board: nxp_lpc1768; framework: a
rduino) 

 
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x > NXP LPC1768`

HARDWARE: LPC1768 100MHz 31.80KB RAM (464KB Flash)
DEBUG: CURRENT(cmsis-dap) ON-BOARD(cmsis-dap) EXTERNAL(blackmagic, jlink)
Converting Marlin.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(off) COMPATIBILITY(strict)
Collected 7 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4
|-- <TMCStepper> 0.2.3
upload disk:  J:
Compiling .pioenvs\LPC1768\src\src\module\delta.cpp.o
Compiling .pioenvs\LPC1768\src\src\module\probe.cpp.o
Compiling .pioenvs\LPC1768\src\src\module\scara.cpp.o
Compiling .pioenvs\LPC1768\src\src\module\servo.cpp.o
Marlin\src\module\delta.cpp: In function 'void home_delta()':
Marlin\src\module\delta.cpp:225:46: error: 'stepperX' was not declared in this scope
stealth_states.x = tmc_enable_stallguard(stepperX);
^~~~~~~~
Marlin\src\module\delta.cpp:225:46: note: suggested alternative: 'strupr'

stealth_states.x = tmc_enable_stallguard(stepperX);
^~~~~~~~
strupr
Marlin\src\module\delta.cpp:226:46: error: 'stepperY' was not declared in this scope
stealth_states.y = tmc_enable_stallguard(stepperY);
^~~~~~~~
Marlin\src\module\delta.cpp:226:46: note: suggested alternative: 'strupr'
stealth_states.y = tmc_enable_stallguard(stepperY);
^~~~~~~~
strupr
Marlin\src\module\delta.cpp:227:46: error: 'stepperZ' was not declared in this scope
stealth_states.z = tmc_enable_stallguard(stepperZ);
^~~~~~~~
Marlin\src\module\delta.cpp:227:46: note: suggested alternative: 'strupr'
stealth_states.z = tmc_enable_stallguard(stepperZ);
^~~~~~~~
strupr
*** [.pioenvs\LPC1768\src\src\module\delta.cpp.o] Error 1
 [ERROR] Took 12.27 seconds
 [SUMMARY]

Environment megaatmega2560              [SKIP]
Environment megaatmega1280              [SKIP]
Environment at90usb1286_cdc             [SKIP]
Environment at90usb1286_dfu             [SKIP]
Environment DUE                         [SKIP]
Environment DUE_USB                     [SKIP]
Environment DUE_debug                   [SKIP]
Environment LPC1769                     [SKIP]
Environment melzi                       [SKIP]
Environment melzi_optiboot              [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment sanguino_atmega1284p        [SKIP]
Environment STM32F1                     [SKIP]
Environment STM32F4                     [SKIP]
Environment ARMED                       [SKIP]
Environment teensy35                    [SKIP]
Environment malyanm200                  [SKIP]
Environment esp32                       [SKIP]
Environment fysetc_f6_13                [SKIP]
Environment LPC1768                     [ERROR]
 [ERROR] Took 12.27 seconds

HolgerSc, 2018-12-27:
go to marlin/src/feature/delta.cpp
add following line:
#include "stepper.h"

vilsed, 2018-12-28:
Fixed! Thank you. And such a simple error too...

closed 2018-12-28 by vilsed