MarlinFirmware/Marlin #12734: Thermal protection

Thowra, 2018-12-26:
### Description

hi,
I wanted to test Marlin 1.1.8 overheating protection.
I put the hotbed sensor to the hotend (I disconnected hotbed heater wire).
I've set up and start heating.
Hotend temperature 185 °
Hotbed temperature 70 °
The temperature increased significantly, but I did not get an error message (185° ).

I configured it:
`#define BED_MAXTEMP 100'`
`----`
`#define THERMAL_PROTECTION_HOTENDS // Enable thermal protection for all extruders`
`#define THERMAL_PROTECTION_BED // Enable thermal protection for the heated bed`
`----`
`#if ENABLED(THERMAL_PROTECTION_HOTENDS)`
`#define THERMAL_PROTECTION_PERIOD 3 //60 Seconds`
`#define THERMAL_PROTECTION_HYSTERESIS 10 // Degrees Celsius`
`----`
`#if ENABLED(THERMAL_PROTECTION_BED)`
`#define THERMAL_PROTECTION_BED_PERIOD 60` 

I use initialize EEPROM after firmware update.
I tried 1.1.8 and 1.1.9.
I use Arduino mega (Atmega2560) and Ramps 1.4.
What might be the problem?
Thanks for your response.

RudyBenner, 2018-12-26:
For bugfix 2.0 the default values are: 
/**
 * Thermal Protection parameters for the bed are just as above for hotends.
 */
#if ENABLED(THERMAL_PROTECTION_BED)
  #define THERMAL_PROTECTION_BED_PERIOD 20    // Seconds
  #define THERMAL_PROTECTION_BED_HYSTERESIS 2 // Degrees Celsius

still the same as Marlin 1.1.8

you did not attach files.

boelle, 2019-02-21:
@Thowra problem solved in bugfix 2.0?

Thowra, 2019-02-21:
@boelle 
Unfortunately, not yet.
So far, I couldn't assign a physical button to preheating.

boelle, 2019-02-21:
and you have tried bugfix 2.0?

is this more a question than a bug?

Thowra, 2019-02-25:
@boelle 
I've tried it, 2.0 is also working badly.
I think this is a mistake (head or bed can overheat).
I tried to solve it with a thermal switchbut I can't insert it as a "push button" to the Marlin.
I learning the "C" language, I working bit slow.

boelle, 2019-02-25:
you did a pid tune right?

Thowra, 2019-02-25:
Yes, I did.

boelle, 2019-02-25:
please attach 2 config files in a zip file

Thowra, 2019-02-25:
I can send the files at the weekend.

boelle, 2019-03-06:
you forgot the files

Thowra, 2019-03-06:
@boelle 
I have not forgotten.
Because of my job I was not a printer near.
I try to solve things at work as soon as possible.

Thowra, 2019-03-09:
@boelle 
These are the last configuration files.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/2948902/config.zip)

boelle, 2019-06-22:
we need a label on this one

Thowra, 2019-06-22:
The problem was only partially solved.

shitcreek, 2019-09-17:
Not sure I understand you correctly, but are you trying to trigger the thermal protection of the bed by heating it with the hotend and not the bed itself? If so, that would not work as the thermal protection of the bed is turned on when its heater is turned on. Same thing with the hotend.

shitcreek, 2019-10-10:
House cleaning. LMK if issue still exists and thread needs to be reopened.

closed 2019-10-10 by shitcreek