MarlinFirmware/Marlin #12736: anet a8 restarts when buttons are pressed

AngryCarrot789, 2018-12-26:
whenever i click any button on the lcd board the printer restarts, whether it's the back button, up/down buttons or enter button.
is there any way to fix this??

Ludy87, 2018-12-26:
check the cable connection from the display

AngryCarrot789, 2018-12-26:
every connector is in correctly... still does it

AngryCarrot789, 2018-12-26:
where do i get those?

AngryCarrot789, 2018-12-26:
also i dont yet have an SD card, i used repetier host over USB to print things, but i accidentally ripped the USB port off the main board. when i removed the whole connector and started it, it started restarting every time i pressed some buttons. i've tried cleaning the port area incase of shorts and looking for blown caps but there's nothing

Ludy87, 2018-12-26:
Okay, you have the stock firmware, is that right?

Then you should deal with the flash of Marlin to rule out a software error.

AngryCarrot789, 2018-12-26:
(saying i did have an SD card avaliable), i cant do anything with the LCD/buttons because it just restarts. how would i force it to reinstall the firmware?

Ludy87, 2018-12-26:
Flash works via Arduino IDE or PlatformIO
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x#building-marlin-20

AngryCarrot789, 2018-12-26:
i'll just buy a new mainboard :)
would be much easier than having to solder wires onto the usb connector pads and trying to make all that work.

trivalik, 2018-12-27:
Building marin and program via the usb cable, does not need any solder. So give it a try for later.

AnHardt, 2018-12-27:
@triffid 
If you eant to support someone - read his messages.

> ripped the USB port off the main board

triffid, 2018-12-28:
@anhardt tagged the wrong person :P

boelle, 2019-02-21:
@AngryCarrot789 problem solved?

closed 2019-03-03 by thinkyhead