MarlinFirmware/Marlin #12742: MKS Gen L board with extruder fan on D7 port not working - how to configure?

I3dee, 2018-12-27:
Hi together,

I have just upgraded my Geeetech i3 Prusa clone to an MKS Gen L board.
Everything works fine but the extruder fan doesn't start on itself.
My setup:
-1 hot end
-Controller fan on 12V port, part cooling fan on D9 to be controlled by slicer, extruder fan on D7 (with mosfet, intended for second hot end) should start at 50°C
- New Marlin 1.1.9

I have found a couple of threads on this topics but wasn't able to solve my problem with the information given there. I found that the pin and fan designation is done as well in the config files as in the pins_ files but I didn't get the settings right.
Can someone help me with this problem? I can post my files if necessary.

Bob

Ludy87, 2018-12-27:
Post your Files (configuration.h configuration_adv.h pins_*.h)

I3dee, 2018-12-27:
[Bobs Marlin config.zip](https://github.com/MarlinFirmware/Marlin/files/2712096/Bobs.Marlin.config.zip)

Ludy87, 2018-12-27:
Kontrolliere die Zeile 242, da fehlt ein * (Sternchen), in der Configuration_adv.h

I3dee, 2018-12-27:
Hallo Ludy,

vielen Dank für Deine schnelle Hilfe- das wars! Ein Sternchen!

Ich finde es echt klasse, dass Ihr hier bei Problemen so unkompliziert helft, so kann auch jemand, der nur knapp die nötigen Kenntnisse hat, mit dem 3D-Druck zurande kommen.

Viele Grüße,

Bob

RudyBenner, 2018-12-27:
Check the output with a multimeter. The fan will not run in reverse. I made this mistake already.

I3dee, 2018-12-27:
Hi Rudy,
thank you for thinking about my problem! It has already been solved by Ludy detecting the missing asterisk. The missing asterisk made my definitions to comments and so they were not used though they were right.

Were polarity matters a lot with this board is the supply connector: + and - are inverted here!

Bob

closed 2018-12-27 by I3dee

kwiatufpv, 2018-12-27:
Pl
99
I Olowqoqqqqqwwl
L1ql3lo1oqqlqoolowl

Ooo
oqomqp