MarlinFirmware/Marlin #12743: Marlin 2.0 and GTM32 temperature issue

PvvS, 2018-12-27:
I managed to translate and install. But the problem is with temperatures. After cooling down, the LCD is set to 0 at the hotend i heatbed. After warming up, the temperature is correct. At least, according to the multimeter.

boelle, 2019-02-21:
@PvvS problem solved?

PvvS, 2019-02-22:
I have to test the last version on a weekend, but it is not yet. After cooling, it shows 0 degrees. After warming up the temperature.

boelle, 2019-03-06:
did you do any further testing?

PvvS, 2019-03-06:
Last weekend I downloaded the last build and the problem persists. Still the same symptoms. At room temperature it measures 0, the temperature is the same as the measured temperature.

boelle, 2019-03-06:
could i ask that you add [BUG] in the title line so it becomes: [BUG]Marlin 2.0 and GTM32 temperature issue

and also attach your 2 config files as a zip file so that @thinkyhead can have a look

closed 2019-07-11 by thinkyhead