MarlinFirmware/Marlin #12744: Connecting" to "Error: EEPROM datasize error.

Glod76, 2018-12-28:
Connecting to: /dev/ttyUSB0
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x66c4f9f0, open=True>(port='/dev/ttyUSB0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: start
Recv: echo:Marlin bugfix-2.0.x
Recv: 
Send: N0 M110 N0*125
Recv: echo: Last Updated: 2018-01-20 | Author: (Donnie Roszel, FT-5 R2 Chimera)
Recv: echo:Compiled: Dec 27 2018
Recv: echo: Free Memory: 2923  PlannerBufferBytes: 1632
Recv: Error:EEPROM datasize error.
Changing monitoring state from "Connecting" to "Error: EEPROM datasize error."
Changing monitoring state from "Error: EEPROM datasize error." to "Offline (Error: EEPROM datasize error.)"
Connection closed, closing down monitor
-->

### Description
Octoprint cant connect
### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

did M502 M500 and a restart
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2713723/Marlin.zip)

Ludy87, 2018-12-28:
```
Error:Field home_offset mismatch.
[ERROR] Error:Field home_offset mismatch.
```

Glod76, 2018-12-29:
still getting the error after new build.

Connecting to: /dev/ttyUSB0
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x6c852ef0, open=True>(port='/dev/ttyUSB0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: start
Send: N0 M110 N0*125
Recv: echo:Marlin bugfix-2.0.x
Recv: 
Recv: echo: Last Updated: 2018-01-20 | Author: (Donnie Roszel, FT-5 R2 Chimera)
Recv: echo:Compiled: Dec 29 2018
Recv: echo: Free Memory: 2923  PlannerBufferBytes: 1632
Recv: Error:EEPROM datasize error.
Changing monitoring state from "Connecting" to "Error: EEPROM datasize error."
Changing monitoring state from "Error: EEPROM datasize error." to "Offline (Error: EEPROM datasize error.)"
Connection closed, closing down monitor

Glod76, 2018-12-30:
Is this fixed yet?

thinkyhead, 2018-12-30:
Patches have been made to the EEPROM code to fix some issues. You should give the current `bugfix-2.0.x` branch a test and see if it has been resolved.

Note that any time you upgrade the firmware, you may get an EEPROM error. If your host is configured to disconnect any time it sees an "Error" it's recommended to disable that feature. EEPROM errors are recoverable.

boelle, 2019-02-21:
@Glod76 problem solved with latest bugfix 2.0?

closed 2019-03-03 by thinkyhead

G-Pereira, 2019-03-03:
Happening again with bugfix-2.0.x https://github.com/MarlinFirmware/Marlin/commit/d71dc5cf42c0055a553ec14d1a400a8e02c334ec

boelle, 2019-03-03:
suggeastion from me is to open new issue

remember to attach your 2 config files as a zip file

thinkyhead, 2019-03-05:
Standard troubleshooting procedure:

- Try `M502` followed by `M500` and reboot
- Add `#define DEBUG_EEPROM_READWRITE` to your config, flash, and test
- Report the additional error output in your next reply

The `DEBUG_EEPROM_READWRITE` option provides more error-checking that we can use to locate potential errors in the EEPROM code.

G-Pereira, 2019-03-05:
Yes, problem solved! I've added an extruder to my printer so the expected EEPROM information size was larger than it actually was, due to new configs for the second extruder. Thanks!

thinkyhead, 2019-03-06:
I'm glad to hear it was a simple fix! A helpful tip is to copy the output from `M503 S0` before you re-flash your board. And then you can send that g-code to the machine after the update to restore your old settings. This is usually safe to do within minor version updates, though some setting values can change meaning across major versions, as with the update from Linear Advance 1.0 - 1.5, or the 50% reduction in jerk values (i.e., 10mm/s instead of 20mm/s) from around version 1.1.6.