MarlinFirmware/Marlin #12745: 1.1.9 - S_CURVE_ACCELERATION does not do anything.

dulfe, 2018-12-28:
I just tried the new `S_CURVE_ACCELERATION `option in the 1.1.9 release and it does not seem to do anything.

The movements with that enabled and disables are exactly the same. Is it working in the 1.1.x branch?

This is the configuration file with `S_CURVE_ACCELERATION `enabled.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2713895/Configuration.zip)

How I tested it?

After uploading the firmware I ran:

```
G28 X
G1 X220 F6000
G1 X0
```

I might be understanding incorrectly how it is supposed to work, but I was expecting to see the movement slowing increasing on start and decreasing at the end of a line.

Any help would be appreciate it.

Thank you.

RudyBenner, 2018-12-28:
* See https://github.com/synthetos/TinyG/wiki/Jerk-Controlled-Motion-Explained

dulfe, 2018-12-28:
@RudyBenner Thank you, I read it and indeed there should be a difference in movements when S_CURVE is enabled and when it is disabled... one is trapezoidal and the other is s-curve.

I tested it and they look exactly the same.... so... I am not testing it correctly (which I am pretty sure it the problem), I set it up incorrectly or it is not working.

Any pointer would be appreciate it.

kAdonis, 2018-12-28:
You can't see the difference with the naked eye, this is normal
but there is a difference, look at #12491