MarlinFirmware/Marlin #12749: Error:EEPROM datasize error.

WrQth, 2018-12-28:
### Description

Unable to use EEPROM with Azteeg X5 mini wifi, Error:EEPROM datasize error. Enabling "#define DEBUG_EEPROM_READWRITE" gives error "Marlin/src/module/configuration_store.cpp:696:19: error: 'SettingsData {aka struct SettingsDataStruct}' has no member named 'lcd_preheat_hotend_temp'; did you mean 'ui_preheat_hotend_temp'?" during build. No LCD installed just onboard SDcard.

### Steps to Reproduce

1. Flashed latest version 2.0.x
2. Multiple SD cards used

**Expected behavior:** M500 save settings

**Actual behavior:** Settings from M500 not saved and receive error

#### Additional Information
[Config_files.zip](https://github.com/MarlinFirmware/Marlin/files/2715395/Config_files.zip)

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

GMagician, 2018-12-28:
Maybe same issue than #12744?

WrQth, 2018-12-28:
#12746 fixed compile but now I get a different error from M500 which is "Error:Field home_offset mismatch"

GMagician, 2018-12-28:
try replace line 490 in configuration_store from:
```c++
dummy = float(DEFAULT_EJERK);
EEPROM_WRITE(dummy);
```

to:
```c++
const float max_jerk[XYZE] = { 10, 10, 0.3, DEFAULT_EJERK };
EEPROM_WRITE(max_jerk);
```

WrQth, 2018-12-28:
Same error "Error:Field home_offset mismatch". I have Junction Deviation enabled so classic jerk isn't being used and if I am understanding the logic of that IF statement then the change made to the code isn't processed.

GMagician, 2018-12-28:
@Ludy87 I checked code where it read data and it seems that in else it use a fro cycle. Am I wrong?

GMagician, 2018-12-28:
@thinkyhead, @Ludy87 I think that #12720 has added this issue.
If jerk data is not saved in `save` function it must neighter be loaded in `load`.
Also it introduced a sanitycheck bug that I fixed with an already posted PR

Ludy87, 2018-12-28:
@GMagician I do not understand why that does not work.
Jerk is disable everywhere if `JUNCTION_DEVIATION` is enable.

GMagician, 2018-12-29:
@Ludy87 you have to look to read function in correspondent `#else`. There is a for cycle that skip "dummy".
If you don't write such dummy values you also need to "not" skip them as well

Nuttavoot, 2018-12-30:
I also get "EEPROM datasize error" when issue M500, M501, M504 on RE-ARM board and no file eeprom.data was created on the SD card.

Nuttavoot, 2019-01-01:
EEPROM commands work correctly for me as well for that build.

boelle, 2019-02-21:
@WrQth problem solved?

WrQth, 2019-02-21:
After rollback in DEC and since then on newer builds it has been working

closed 2019-02-21 by WrQth